COVESA/vsomeip #276: reactor_op_ext missing parameter with boost helpers 1.74

kheaactua, 2021-08-24:
In vsomeip 3.1.20.3, the 1.74 boost helps appear to be missing a parameter to [reactor_op_ext](https://github.com/GENIVI/vsomeip/blob/13f9c89ced6ffaeb1faf485152e27e1f40d234cd/implementation/helper/1.74/boost/asio/detail/reactive_socket_recvfrom_op_ext.hpp#L44) and [reactor_op_ext_local](https://github.com/GENIVI/vsomeip/blob/13f9c89ced6ffaeb1faf485152e27e1f40d234cd/implementation/helper/1.74/boost/asio/detail/reactive_socket_recvfrom_op_ext_local.hpp#L40).

Adding the `success_ec` parameter first fixes the compilation error.

I encountered this when building natively in AOSP - I don't encounter it in my QNX (7.1 gcc 8.3) builds and I have no idea why yet (still working through the code.)

yunfeiyangcc, 2021-11-09:
Yes, I have the same problem on Android. @lutzbichler Is there any way to avoid it ?

kheaactua, 2021-11-09:
I patched it in our local fork.....

I'd be happy to submit a PR if there's any change genivi would accept it.

goncaloalmeida, 2023-03-15:
Fixed on version 3.3.0 (master branch).

closed 2023-03-15 by goncaloalmeida