COVESA/vsomeip #277: How to configure services in different VLAN channels

ZhangHuarong-CN, 2021-08-25:
Requirements:
1.The 0x301 service runs on the VLAN2 channel,and its IP address is 162.16.2.97.The service contains an offer and requires client to subscribe.The client's IP is 162.16.2.99.
2.1.The 0x302 service runs on the VLAN3 channel, and its IP address is 162.16.3.97,The service contains an offer and requires client to subscribe.The client's IP is 162.16.3.99.

How to use the vsomeip library to implement blow requriments,These two services on different VLANs.

ByteXiaoTang, 2021-09-09:
use two different configure json

ZhangHuarong-CN, 2021-09-11:
> use two different configure json

Hello Tang,
Thank you,If two json files are used,it means that two vsomeip application processes need to be started,Rightly?

Commments:
I failed the test with this method on vsomeip 2.14.16,error log as blow:
[error] Subscriber's IP isn't in the same subnet as host's IP: 172.16.2.99
[error] Invalid port or IP address in first IPv4 endpoint option specified! 172.16.2.99 session: 0108

ByteXiaoTang, 2021-09-12:
First, you have two clans. Second, although process read different configures and they belong to different vlans(networks in configure file). Proxy still try to connect the different stub, but can just the same  vlan stub will be connected------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年9月11日(星期六) 上午9:50
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [GENIVI/vsomeip] How to configure services in different VLAN channels (#277)

ZhangHuarong-CN, 2021-09-15:
Hello Tang,
After I updated the vsomeip library to the latest version 3.1.20.3,I can realize the multi-host IP sending function,thank you.

ZhangHuarong-CN, 2021-09-15:
This issue have been solved

closed 2021-09-15 by ZhangHuarong-CN

omitheaussie, 2024-10-29:
Hello ZhangHuarong. I have the same issue as you faced
[image]
But I am already on the 3.1.20.3 library version. What were the VSomeIPConfig.json settings that you had to make this work?