COVESA/vsomeip #279: Header in Wireshark

tobsenthomas2, 2021-09-02:
Hi and thx for the great work!
I don't get the mean of a  few Bytes in my Wireshark trace.
[image]

I dont know what the Bytes in the circle should stand for. Can someone help? (maybe there is also a docu i didn't find?)
Where the first 02 stands it is incrementing after each frame. Is there anywhere a config where i can "switch them off"?

tobsenthomas2, 2021-09-03:
Maybe as additional Info:
12 43 80 01 Message ID
00 00 02 40 Length
**00 00 00 02 01 00 02 00 unknown**
12 55 00 03 RequestID
01 00 00 00 Protocol Version, IV, MT and Return Code

closed 2021-09-03 by tobsenthomas2

ziyangfu, 2021-12-28:
Hi， can you using wireshark to parse the someip header？
I can not do it like following image.
wireshark version ： 3.6
[image]