COVESA/vsomeip #280: TCP Tx stuck by 'socket->is_open()' return FALSE

songkilyong, 2021-09-08:
Hello,
I'm using vsomeip 3.1.20.3 now, but I have an issue, 'TCP Tx block'.
When DUT is connected to other device and its SOME/IP service via TCP client port, 
if other device reboot by Power reset,(or other reason)
then DUT's vsomeip detect reboot of other device 
and 'client::connect_cbk()' calls 'shutdown and close socket(true)'.
Then, after the SOME/IP-SD subscription, DUT trying to send SomeIpGetter to get information from other device.

IN THIS SITUATION,
the condition, '**socket_->is_open()' in 'tcp_client_endpoint_impl::send_queued()' returns FALSE, sometimes,**
so the **'async_write()' of boost library cannot be called, then the queued TCP data is remained forever.**
After this, **all of TCP tx attampt are blocked, 
because 'client_endpoint_impl::queue_train()' cannot call 'send_queued()' since queue_.size() != 0**. => **TCP Tx block**

So, I want to ask 2 questions,

1) When 'socket_->is_open()'(line 325, tcp_client_endpoint_impl.cpp) can return FALSE?
: I want to know the reason why is_open() return FALSE, is this because of client port? or server port?
and there is no condition check for 'socket_->is_open() == FALSE' from whole vsomeip source code,
then is that mean, socket_->is_open() always return TRUE?

2) Do you have any correct solution or workaround for this issue?
: I already tested, when socket_->is_open() of tcp_client_endpoint_impl::send_queued() return FALSE,
I just call queue_.pop_front() and reduce queue_size, then the TCP queue works well,
but I want to correct fix from official way.(=without any side effect)

Thank you very much in advance.

ByteXiaoTang, 2021-09-09:
not care . because has been solved before

hanul0505, 2021-09-09:
> not care . because has been solved before

What do you mean 'has been solved before'? I'm using v3.1.20.3, the latest version now and still have problem.

ByteXiaoTang, 2021-09-13:
> > not care . because has been solved before
> 
> What do you mean 'has been solved before'? Do you have any information for fixed issue?
> I'm using v3.1.20.3, the latest version, and the other device is using v3.1.15.1.

sorry , i may miss something.

closed 2023-06-07 by fcmonteiro