COVESA/vsomeip #281: create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type

swendt78, 2021-09-13:
Hi, we are facing a similar issue as https://github.com/GENIVI/vsomeip/issues/229
We have configured our system to work with TCP but somehow attribute notifications do not work.
Do we have to use UDP for notifications ?

Please find attached our config files for CommonAPI and SomeIP

[FIDL_FDEPL_JSON.zip](https://github.com/GENIVI/vsomeip/files/7155075/FIDL_FDEPL_JSON.zip)

This is the log of the server:
16:18:24: Starte /usr/local/bin/RadioService ...
2021-01-02 15:31:51.386848 [info] Parsed vsomeip configuration in 5ms
2021-01-02 15:31:51.390243 [info] Using configuration file: "/etc/vsomeip.json".
2021-01-02 15:31:51.392328 [info] Configuration module loaded.
2021-01-02 15:31:51.392823 [info] Initializing vsomeip application "".
2021-01-02 15:31:51.393145 [info] Instantiating routing manager [Proxy].
2021-01-02 15:31:51.394458 [info] Client [ffff] is connecting to [0] at /tmp/vsomeip-0
2021-01-02 15:31:51.395066 [info] Application(unnamed, ffff) is initialized (11, 100).
2021-01-02 15:31:51.401637 [info] Starting vsomeip application "" (ffff) using 2 threads I/O nice 255
2021-01-02 15:31:51.404226 [info] main dispatch thread id from application: ffff () is: ffff95809050 TID: 775
2021-01-02 15:31:51.408422 [info] shutdown thread id from application: ffff () is: ffff95008050 TID: 776
2021-01-02 15:31:51.409151 [info] io thread id from application: ffff () is: ffff9600a050 TID: 774
2021-01-02 15:31:51.410111 [info] io thread id from application: ffff () is: ffff94807050 TID: 777
2021-01-02 15:31:51.414902 [info] Listening at /tmp/vsomeip-101
2021-01-02 15:31:51.416187 [info] Client 101 () successfully connected to routing  ~> registering..
2021-01-02 15:31:51.431767 [info] Application/Client 101 () is registered.
Could not open gpio value file for IO: 65 in SysFS2021-01-02 15:31:51.436678 [info] REGISTER EVENT(0101): [1234.5678.80ea:is_provider=1]
2021-01-02 15:31:51.439326 [info] REGISTER EVENT(0101): [1234.5678.80ec:is_provider=1]
2021-01-02 15:31:51.439947 [info] REGISTER EVENT(0101): [1234.5678.80ee:is_provider=1]
2021-01-02 15:31:51.440302 [info] REGISTER EVENT(0101): [1234.5678.80f0:is_provider=1]
2021-01-02 15:31:51.440668 [info] REGISTER EVENT(0101): [1234.5678.80f2:is_provider=1]
2021-01-02 15:31:51.441042 [info] REGISTER EVENT(0101): [1234.5678.80f4:is_provider=1]
2021-01-02 15:31:51.441408 [info] REGISTER EVENT(0101): [1234.5678.80f6:is_provider=1]
2021-01-02 15:31:51.441773 [info] REGISTER EVENT(0101): [1234.5678.80f8:is_provider=1]
2021-01-02 15:31:51.442142 [info] REGISTER EVENT(0101): [1234.5678.80fa:is_provider=1]
2021-01-02 15:31:51.442501 [info] REGISTER EVENT(0101): [1234.5678.80fc:is_provider=1]
2021-01-02 15:31:51.442862 [info] REGISTER EVENT(0101): [1234.5678.80fe:is_provider=1]
2021-01-02 15:31:51.443227 [info] REGISTER EVENT(0101): [1234.5678.8100:is_provider=1]
2021-01-02 15:31:51.443595 [info] REGISTER EVENT(0101): [1234.5678.8102:is_provider=1]
2021-01-02 15:31:51.443971 [info] REGISTER EVENT(0101): [1234.5678.8104:is_provider=1]
2021-01-02 15:31:51.444342 [info] REGISTER EVENT(0101): [1234.5678.8106:is_provider=1]
2021-01-02 15:31:51.444671 [info] REGISTER EVENT(0101): [1234.5678.8108:is_provider=1]
2021-01-02 15:31:51.445006 [info] REGISTER EVENT(0101): [1234.5678.810a:is_provider=1]
2021-01-02 15:31:51.445377 [info] REGISTER EVENT(0101): [1234.5678.810c:is_provider=1]
2021-01-02 15:31:51.445742 [info] REGISTER EVENT(0101): [1234.5678.810e:is_provider=1]
2021-01-02 15:31:51.446106 [info] REGISTER EVENT(0101): [1234.5678.8110:is_provider=1]
2021-01-02 15:31:51.446463 [info] REGISTER EVENT(0101): [1234.5678.814c:is_provider=1]
2021-01-02 15:31:51.446821 [info] REGISTER EVENT(0101): [1234.5678.814d:is_provider=1]

And this is the output of the client:
2021-08-13 16:18:31.876762 [info] Parsed vsomeip configuration in 0ms
2021-08-13 16:18:31.877162 [info] Using configuration file: "/tmp/vsomeipclient.json".
2021-08-13 16:18:31.877260 [info] Configuration module loaded.
2021-08-13 16:18:31.877310 [info] Initializing vsomeip application "ServicesDemo".
2021-08-13 16:18:31.877440 [info] Instantiating routing manager [Host].
2021-08-13 16:18:31.878208 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2021-08-13 16:18:31.878824 [info] Service Discovery enabled. Trying to load module.
2021-08-13 16:18:31.881476 [info] Service Discovery module loaded.
2021-08-13 16:18:31.881767 [info] Application(ServicesDemo, 1314) is initialized (11, 100).
2021-08-13 16:18:31.882982 [info] REGISTER EVENT(1314): [1234.5678.80ea:is_provider=false]
2021-08-13 16:18:31.883167 [info] REGISTER EVENT(1314): [1234.5678.80ec:is_provider=false]
2021-08-13 16:18:31.882997 [info] Starting vsomeip application "ServicesDemo" (1314) using 2 threads I/O nice 255
2021-08-13 16:18:31.883270 [info] REGISTER EVENT(1314): [1234.5678.80ee:is_provider=false]
2021-08-13 16:18:31.883511 [info] REGISTER EVENT(1314): [1234.5678.80f0:is_provider=false]
2021-08-13 16:18:31.883604 [info] REGISTER EVENT(1314): [1234.5678.80f2:is_provider=false]
2021-08-13 16:18:31.883680 [info] REGISTER EVENT(1314): [1234.5678.80f4:is_provider=false]
2021-08-13 16:18:31.883753 [info] REGISTER EVENT(1314): [1234.5678.80f6:is_provider=false]
2021-08-13 16:18:31.883837 [info] REGISTER EVENT(1314): [1234.5678.80f8:is_provider=false]
2021-08-13 16:18:31.883908 [info] REGISTER EVENT(1314): [1234.5678.80fa:is_provider=false]
2021-08-13 16:18:31.884014 [info] REGISTER EVENT(1314): [1234.5678.80fc:is_provider=false]
2021-08-13 16:18:31.883919 [info] shutdown thread id from application: 1314 (ServicesDemo) is: 7f7d9bfff700 TID: 3233859
[... 142 lines cut ...]
2021-08-13 16:18:32.892388 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.816c] 2
2021-08-13 16:18:32.892478 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.816e] 2
2021-08-13 16:18:32.892561 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8170] 2
2021-08-13 16:18:32.892647 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8172] 2
2021-08-13 16:18:32.892731 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8174] 2
2021-08-13 16:18:34.307400 [error] Routing info for remote service could not be found! (1314): [1234.5678.0bd6] 0001
2021-08-13 16:18:34.894331 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.814e] 2
2021-08-13 16:18:34.895117 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8150] 2
2021-08-13 16:18:34.895329 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8152] 2
2021-08-13 16:18:34.895498 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8154] 2
2021-08-13 16:18:34.895672 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8156] 2
2021-08-13 16:18:34.895834 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8158] 2
2021-08-13 16:18:34.896002 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.815a] 2
2021-08-13 16:18:34.896171 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.815c] 2
2021-08-13 16:18:34.896329 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.815e] 2
2021-08-13 16:18:34.896489 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8162] 2
2021-08-13 16:18:34.896653 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8164] 2
2021-08-13 16:18:34.896812 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8166] 2
[... cut ...]
2021-08-13 16:18:34.897456 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8170] 2
2021-08-13 16:18:34.897618 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8172] 2
2021-08-13 16:18:34.897779 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8174] 2
2021-08-13 16:18:36.900012 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.814e] 2

Sudozh, 2021-12-18:
> 使用的vsomeip和CommonAPI的版本分别是3.1.20, 3.2.0吗？在这个版本上我也遇到了同样的问题。 “reliability type: [1234.5678.814e] 2” 表示使用的是TCP通信协议。 在json配置文件中改成UDP就能正常通信了。（如果报文太大，记得配置分包）

想请问一下，需要修改的是JSON的哪个配置文件呢？

Sudozh, 2021-12-21:
I also encountered the same problem, have you solved this problem?

UItogether, 2022-07-18:
> 使用的vsomeip和CommonAPI的版本分别是3.1.20, 3.2.0吗？在这个版本上我也遇到了同样的问题。 “reliability type: [1234.5678.814e] 2” 表示使用的是TCP通信协议。 在json配置文件中改成UDP就能正常通信了。（如果报文太大，记得配置分包）

你好请问，如何利用someip进行拆包和组包，有案例可以学习吗，非常感谢

mohitrao12, 2022-12-06:
> Hi, we are facing a similar issue as #229 We have configured our system to work with TCP but somehow attribute notifications do not work. Do we have to use UDP for notifications ?
> 
> Please find attached our config files for CommonAPI and SomeIP
> 
> [FIDL_FDEPL_JSON.zip](https://github.com/GENIVI/vsomeip/files/7155075/FIDL_FDEPL_JSON.zip)
> 
> This is the log of the server: 16:18:24: Starte /usr/local/bin/RadioService ... 2021-01-02 15:31:51.386848 [info] Parsed vsomeip configuration in 5ms 2021-01-02 15:31:51.390243 [info] Using configuration file: "/etc/vsomeip.json". 2021-01-02 15:31:51.392328 [info] Configuration module loaded. 2021-01-02 15:31:51.392823 [info] Initializing vsomeip application "". 2021-01-02 15:31:51.393145 [info] Instantiating routing manager [Proxy]. 2021-01-02 15:31:51.394458 [info] Client [ffff] is connecting to [0] at /tmp/vsomeip-0 2021-01-02 15:31:51.395066 [info] Application(unnamed, ffff) is initialized (11, 100). 2021-01-02 15:31:51.401637 [info] Starting vsomeip application "" (ffff) using 2 threads I/O nice 255 2021-01-02 15:31:51.404226 [info] main dispatch thread id from application: ffff () is: ffff95809050 TID: 775 2021-01-02 15:31:51.408422 [info] shutdown thread id from application: ffff () is: ffff95008050 TID: 776 2021-01-02 15:31:51.409151 [info] io thread id from application: ffff () is: ffff9600a050 TID: 774 2021-01-02 15:31:51.410111 [info] io thread id from application: ffff () is: ffff94807050 TID: 777 2021-01-02 15:31:51.414902 [info] Listening at /tmp/vsomeip-101 2021-01-02 15:31:51.416187 [info] Client 101 () successfully connected to routing ~> registering.. 2021-01-02 15:31:51.431767 [info] Application/Client 101 () is registered. Could not open gpio value file for IO: 65 in SysFS2021-01-02 15:31:51.436678 [info] REGISTER EVENT(0101): [1234.5678.80ea:is_provider=1] 2021-01-02 15:31:51.439326 [info] REGISTER EVENT(0101): [1234.5678.80ec:is_provider=1] 2021-01-02 15:31:51.439947 [info] REGISTER EVENT(0101): [1234.5678.80ee:is_provider=1] 2021-01-02 15:31:51.440302 [info] REGISTER EVENT(0101): [1234.5678.80f0:is_provider=1] 2021-01-02 15:31:51.440668 [info] REGISTER EVENT(0101): [1234.5678.80f2:is_provider=1] 2021-01-02 15:31:51.441042 [info] REGISTER EVENT(0101): [1234.5678.80f4:is_provider=1] 2021-01-02 15:31:51.441408 [info] REGISTER EVENT(0101): [1234.5678.80f6:is_provider=1] 2021-01-02 15:31:51.441773 [info] REGISTER EVENT(0101): [1234.5678.80f8:is_provider=1] 2021-01-02 15:31:51.442142 [info] REGISTER EVENT(0101): [1234.5678.80fa:is_provider=1] 2021-01-02 15:31:51.442501 [info] REGISTER EVENT(0101): [1234.5678.80fc:is_provider=1] 2021-01-02 15:31:51.442862 [info] REGISTER EVENT(0101): [1234.5678.80fe:is_provider=1] 2021-01-02 15:31:51.443227 [info] REGISTER EVENT(0101): [1234.5678.8100:is_provider=1] 2021-01-02 15:31:51.443595 [info] REGISTER EVENT(0101): [1234.5678.8102:is_provider=1] 2021-01-02 15:31:51.443971 [info] REGISTER EVENT(0101): [1234.5678.8104:is_provider=1] 2021-01-02 15:31:51.444342 [info] REGISTER EVENT(0101): [1234.5678.8106:is_provider=1] 2021-01-02 15:31:51.444671 [info] REGISTER EVENT(0101): [1234.5678.8108:is_provider=1] 2021-01-02 15:31:51.445006 [info] REGISTER EVENT(0101): [1234.5678.810a:is_provider=1] 2021-01-02 15:31:51.445377 [info] REGISTER EVENT(0101): [1234.5678.810c:is_provider=1] 2021-01-02 15:31:51.445742 [info] REGISTER EVENT(0101): [1234.5678.810e:is_provider=1] 2021-01-02 15:31:51.446106 [info] REGISTER EVENT(0101): [1234.5678.8110:is_provider=1] 2021-01-02 15:31:51.446463 [info] REGISTER EVENT(0101): [1234.5678.814c:is_provider=1] 2021-01-02 15:31:51.446821 [info] REGISTER EVENT(0101): [1234.5678.814d:is_provider=1]
> 
> And this is the output of the client: 2021-08-13 16:18:31.876762 [info] Parsed vsomeip configuration in 0ms 2021-08-13 16:18:31.877162 [info] Using configuration file: "/tmp/vsomeipclient.json". 2021-08-13 16:18:31.877260 [info] Configuration module loaded. 2021-08-13 16:18:31.877310 [info] Initializing vsomeip application "ServicesDemo". 2021-08-13 16:18:31.877440 [info] Instantiating routing manager [Host]. 2021-08-13 16:18:31.878208 [info] create_local_server Routing endpoint at /tmp/vsomeip-0 2021-08-13 16:18:31.878824 [info] Service Discovery enabled. Trying to load module. 2021-08-13 16:18:31.881476 [info] Service Discovery module loaded. 2021-08-13 16:18:31.881767 [info] Application(ServicesDemo, 1314) is initialized (11, 100). 2021-08-13 16:18:31.882982 [info] REGISTER EVENT(1314): [1234.5678.80ea:is_provider=false] 2021-08-13 16:18:31.883167 [info] REGISTER EVENT(1314): [1234.5678.80ec:is_provider=false] 2021-08-13 16:18:31.882997 [info] Starting vsomeip application "ServicesDemo" (1314) using 2 threads I/O nice 255 2021-08-13 16:18:31.883270 [info] REGISTER EVENT(1314): [1234.5678.80ee:is_provider=false] 2021-08-13 16:18:31.883511 [info] REGISTER EVENT(1314): [1234.5678.80f0:is_provider=false] 2021-08-13 16:18:31.883604 [info] REGISTER EVENT(1314): [1234.5678.80f2:is_provider=false] 2021-08-13 16:18:31.883680 [info] REGISTER EVENT(1314): [1234.5678.80f4:is_provider=false] 2021-08-13 16:18:31.883753 [info] REGISTER EVENT(1314): [1234.5678.80f6:is_provider=false] 2021-08-13 16:18:31.883837 [info] REGISTER EVENT(1314): [1234.5678.80f8:is_provider=false] 2021-08-13 16:18:31.883908 [info] REGISTER EVENT(1314): [1234.5678.80fa:is_provider=false] 2021-08-13 16:18:31.884014 [info] REGISTER EVENT(1314): [1234.5678.80fc:is_provider=false] 2021-08-13 16:18:31.883919 [info] shutdown thr [... cut ...] s subscription doesn't match reliability type: [1234.5678.8172] 2 2021-08-13 16:18:34.897779 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.8174] 2 2021-08-13 16:18:36.900012 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.814e] 2

I also encountered the same problem, have you solved this problem?
If someone solved then please let me know.

Thank You.

anaritarodrigues, 2023-09-06:
@swendt78, Which version of CommonAPI were you using?

closed 2023-10-02 by goncaloalmeida