COVESA/vsomeip #282: compile issue in AOSP

naocanyouji, 2021-09-27:
when building library libvsomeip3-cfg with android.bp in AOSP, some errors happened, these are error logs, can someone help fix?

ld.lld: error: undefined symbol: boost::filesystem::detail::directory_iterator_construct(boost::filesystem::directory_iterator&, boost::filesystem::path const&, boost::system::error_code*)
>>> referenced by operations.hpp:957 (external/vsomeip/vsomeip-master/external/boost_1_70/include/boost/filesystem/operations.hpp:957)
>>>               out/soong/.intermediates/external/vsomeip/vsomeip-master/libvsomeip3-cfg/android_vendor.30_arm64_armv8-a_shared/obj/external/vsomeip/vsomeip-master/implementation/configuration/src/configuration_impl.o:(vsomeip_v3::cfg::configuration_impl::read_data(std::__1::set<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<vsomeip_v3::configuration_element, std::__1::allocator<vsomeip_v3::configuration_element> >&, std::__1::set<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >&, bool))

nkh-lab, 2021-09-27:
Do you have Boost libraries in your AOSP?

nkh-lab, 2021-12-02:
Like example you can use this AOSP manifest project where vsomeip and boost libraries are built in: [https://github.com/nkh-lab/aosp-ncar-manifest](https://github.com/nkh-lab/aosp-ncar-manifest). Just sync, compile AOSP, compile [vsomeip hello_world example](https://github.com/COVESA/vsomeip/tree/master/examples/hello_world) and run it on emulator.

ByteXiaoTang, 2021-12-10:
Boost not right

ByteXiaoTang, 2021-12-17:
git clone -b boost-1.70.0 git@github.com:boostorg/accumulators.git
git clone -b boost-1.70.0 git@github.com:boostorg/algorithm.git
git clone -b boost-1.70.0 git@github.com:boostorg/align.git
git clone -b boost-1.70.0 git@github.com:boostorg/any.git
git clone -b boost-1.70.0 git@github.com:boostorg/array.git
git clone -b boost-1.70.0 git@github.com:boostorg/asio.git
git clone -b boost-1.70.0 git@github.com:boostorg/assert.git
git clone -b boost-1.70.0 git@github.com:boostorg/assign.git
git clone -b boost-1.70.0 git@github.com:boostorg/atomic.git
git clone -b boost-1.70.0 git@github.com:boostorg/beast.git
git clone -b boost-1.70.0 git@github.com:boostorg/bimap.git
git clone -b boost-1.70.0 git@github.com:boostorg/bind.git
git clone -b boost-1.70.0 git@github.com:boostorg/callable_traits.git
git clone -b boost-1.70.0 git@github.com:boostorg/chrono.git
git clone -b boost-1.70.0 git@github.com:boostorg/circular_buffer.git
git clone -b boost-1.70.0 git@github.com:boostorg/compatibility.git
git clone -b boost-1.70.0 git@github.com:boostorg/compute.git
git clone -b boost-1.70.0 git@github.com:boostorg/concept_check.git
git clone -b boost-1.70.0 git@github.com:boostorg/config.git
git clone -b boost-1.70.0 git@github.com:boostorg/container.git
git clone -b boost-1.70.0 git@github.com:boostorg/container_hash.git
git clone -b boost-1.70.0 git@github.com:boostorg/context.git
git clone -b boost-1.70.0 git@github.com:boostorg/conversion.git
git clone -b boost-1.70.0 git@github.com:boostorg/convert.git
git clone -b boost-1.70.0 git@github.com:boostorg/core.git
git clone -b boost-1.70.0 git@github.com:boostorg/coroutine.git
git clone -b boost-1.70.0 git@github.com:boostorg/coroutine2.git
git clone -b boost-1.70.0 git@github.com:boostorg/crc.git
git clone -b boost-1.70.0 git@github.com:boostorg/date_time.git
git clone -b boost-1.70.0 git@github.com:boostorg/detail.git
git clone -b boost-1.70.0 git@github.com:boostorg/disjoint_sets.git
git clone -b boost-1.70.0 git@github.com:boostorg/dll.git
git clone -b boost-1.70.0 git@github.com:boostorg/dynamic_bitset.git
git clone -b boost-1.70.0 git@github.com:boostorg/endian.git
git clone -b boost-1.70.0 git@github.com:boostorg/exception.git
git clone -b boost-1.70.0 git@github.com:boostorg/fiber.git
git clone -b boost-1.70.0 git@github.com:boostorg/filesystem.git
git clone -b boost-1.70.0 git@github.com:boostorg/flyweight.git
git clone -b boost-1.70.0 git@github.com:boostorg/foreach.git
git clone -b boost-1.70.0 git@github.com:boostorg/format.git
git clone -b boost-1.70.0 git@github.com:boostorg/function.git
git clone -b boost-1.70.0 git@github.com:boostorg/functional.git
git clone -b boost-1.70.0 git@github.com:boostorg/function_types.git
git clone -b boost-1.70.0 git@github.com:boostorg/fusion.git
git clone -b boost-1.70.0 git@github.com:boostorg/geometry.git
git clone -b boost-1.70.0 git@github.com:boostorg/gil.git
git clone -b boost-1.70.0 git@github.com:boostorg/graph.git
git clone -b boost-1.70.0 git@github.com:boostorg/graph_parallel.git
git clone -b boost-1.70.0 git@github.com:boostorg/hana.git
git clone -b boost-1.70.0 git@github.com:boostorg/heap.git
git clone -b boost-1.70.0 git@github.com:boostorg/icl.git
git clone -b boost-1.70.0 git@github.com:boostorg/integer.git
git clone -b boost-1.70.0 git@github.com:boostorg/interprocess.git
git clone -b boost-1.70.0 git@github.com:boostorg/intrusive.git
git clone -b boost-1.70.0 git@github.com:boostorg/io.git
git clone -b boost-1.70.0 git@github.com:boostorg/iostreams.git
git clone -b boost-1.70.0 git@github.com:boostorg/iterator.git
git clone -b boost-1.70.0 git@github.com:boostorg/lambda.git
git clone -b boost-1.70.0 git@github.com:boostorg/lexical_cast.git
git clone -b boost-1.70.0 git@github.com:boostorg/locale.git
git clone -b boost-1.70.0 git@github.com:boostorg/local_function.git
git clone -b boost-1.70.0 git@github.com:boostorg/lockfree.git
git clone -b boost-1.70.0 git@github.com:boostorg/log.git
git clone -b boost-1.70.0 git@github.com:boostorg/logic.git
git clone -b boost-1.70.0 git@github.com:boostorg/math.git
git clone -b boost-1.70.0 git@github.com:boostorg/metaparse.git
git clone -b boost-1.70.0 git@github.com:boostorg/move.git
git clone -b boost-1.70.0 git@github.com:boostorg/mp11.git
git clone -b boost-1.70.0 git@github.com:boostorg/mpi.git
git clone -b boost-1.70.0 git@github.com:boostorg/mpl.git
[... 365 lines cut ...]
    ],
    header_libs: [
        "libboost_headers",
    ],
    shared: {
        shared_libs: [
            "libboost_system",
        ],
    },
    srcs : [
        "thread/src/**/*.cpp",
    ],
    exclude_srcs : [
        "thread/src/win32/*.cpp",
        "thread/src/pthread/once_atomic.cpp",
    ],
    local_include_dirs : [
        "thread/src",
        "thread/src/pthread",
    ],
    export_include_dirs: ["thread/include"],
    export_header_lib_headers: [
        "libboost_headers",
    ],
}

closed 2023-03-03 by BrunoLSilvaCTW