COVESA/vsomeip #284: Type in the documentation.

brenkem, 2021-09-30:
There is one "for" to much: …gister callbacks **for for** messag…

See: 
- second line of _register_message_handler()_
- vsomeip3/build/documentation/html/classvsomeip__v3_1_1application.html#a0f3a2f62fcee8a6a1beba1295c94ac65

sisco0, 2021-11-14:
Could you kindly provide a link to the source code files or the filename and line that is having this issue? As I could not find it by using `grep -iRn "for for"` in the root folder.

brenkem, 2021-11-14:
Hi,

you need to build the documentation first, see the HTML link for the doxygen based documentation above.
I will check again when I am back the the office.

Thanks.

goncaloalmeida, 2023-02-17:
@brenkem this problem still occur?

goncaloalmeida, 2023-02-23:
Closing the ticket -> if the problem still occurs please reopen the ticket.

closed 2023-02-23 by goncaloalmeida