COVESA/vsomeip #289: response-sample routing manager instantiate problem (QNX + ARM64)

yoloisvtyois, 2021-11-02:
Following error is given when trying to run `env VSOMEIP_CONFIGURATION=vsomeip-local.json VSOMEIP_APPLICATION_NAME=service-sample ./response-sample`;
_[error] application: service-sample configured as routing but other routing manager present. Won't instantiate routing
Couldn't initialize application_

no other vsomeip application is running besides response-sample, what may be the problem?

Thanks,

lutzbichler, 2021-12-07:
Does /tmp/vsomeip.lck exist? If yes, remove it.

ByteXiaoTang, 2021-12-10:
Another process run,kill the process firstly

Leo-Jie, 2022-11-03:
I also have this issue in windows 10 OS. But after I modified the VSOMEIP_APPLICATION_NAME different to the routine name in Json file. Then it works but with below error information. Anyone has same issue ?

2022-10-03 17:37:48.848880 [warning] Reliability type for event [1234.5678.777] was not configured Using : RT_UNRELIABLE
2022-10-03 17:37:48.849080 [warning] Reliability type for event [1234.5678.778] was not configured Using : RT_UNRELIABLE
2022-10-03 17:37:48.849243 [warning] Reliability type for event [1234.5678.779] was not configured Using : RT_UNRELIABLE
2022-10-03 17:37:48.849437 [info] Parsed vsomeip configuration in 7ms
2022-10-03 17:37:48.849543 [info] Using configuration file: "C:\ALAP\vsomeip\examples\Debug\vsomeip-local.json".
2022-10-03 17:37:48.849648 [info] Configuration module loaded.
2022-10-03 17:37:48.849754 [info] Initializing vsomeip application "service-sample1".
2022-10-03 17:37:48.849876 [info] Instantiating routing manager [Proxy].
2022-10-03 17:37:48.849954 [info] Connecting to [0] at c822
2022-10-03 17:37:48.850030 [info] Application(service-sample1, ffff) is initialized (11, 100).
Static routing OFF
2022-10-03 17:37:48.850114 [info] Starting vsomeip application "service-sample1" (ffff) using 2 threads
2022-10-03 17:37:48.850195 [info] main dispatch thread id from application: ffff (service-sample1) is: 2b2c
2022-10-03 17:37:48.850198 [info] shutdown thread id from application: ffff (service-sample1) is: 2834
2022-10-03 17:37:49.950412 [info] io thread id from application: ffff (service-sample1) is: 1d34
2022-10-03 17:37:49.950434 [info] io thread id from application: ffff (service-sample1) is: 3b4
2022-10-03 17:37:49.950681 [info] **routing_manager_proxy::on_disconnect: Client 0xffff calling host_->on_state with DEREGISTERED**
Application service-sample1 is deregistered.
2022-10-03 17:37:50.060781 [info] **routing_manager_proxy::on_disconnect: Client 0xffff calling host_->on_state with DEREGISTERED**
Application service-sample1 is deregistered.
2022-10-03 17:37:51.182482 [info] **routing_manager_proxy::on_disconnect: Client 0xffff calling host_->on_state with DEREGISTERED**

Leo-Jie, 2022-11-03:
> Does /tmp/vsomeip.lck exist? If yes, remove it.

Thank you, This solution could fix the issue.

mohitrao12, 2022-12-02:
> I also have this issue in windows 10 OS. But after I modified the VSOMEIP_APPLICATION_NAME different to the routine name in Json file. Then it works but with below error information. Anyone has same [...]

yes i have the same issue RT_UNRELIABLE  if u know how to resolve please let me knoe

closed 2023-03-03 by BrunoLSilvaCTW

AhMmedMahmoud, 2025-02-17:
does anyone know solution for 
[error] application: mE_publisher configured as routing but other routing manager present. Won't instantiate routing

duartenfonseca, 2025-02-25:
@AhMmedMahmoud you have another app being used as routingmanager