COVESA/vsomeip #290: How can I not use UNIX domain socket.

acodercat, 2021-11-05:
Because I want to use tcpdump to capture vsomeip message. 
I can't use tcpdump to capture vsomeip message in UNIX domain socket.

zmurf, 2021-12-02:
Yes. I would also like to know if it is possible to disable UDS and communicate over TCP even on local services.

zhangslice, 2021-12-07:
Have you solved your problem?  I have the same problem, any good advice

lutzbichler, 2021-12-07:
The internal communication always uses Unix domain sockets on Linux and TCP on Windows. It is not possible to change this without changing the code.

zmurf, 2021-12-07:
I solved it by running one of the applications in a docker container with bridged network. But that is often very cumbersome.

I also tried solving it by changing the vsomeip code, and it works... But that is not really a viable solution. We have no wish to add that kind of maintenance to our project.

goncaloalmeida, 2023-02-17:
Closing ticket: Ticket has already been answered, please check the answer:

> The internal communication always uses Unix domain sockets on Linux and TCP on Windows. It is not possible to change this without changing the code.

closed 2023-02-17 by goncaloalmeida