COVESA/vsomeip #291: Missing explanation for 'magic cookies'

andrasi, 2021-11-08:
Hello!

What are the magic cookies used for?

Looking at the documentation and tests I couldn't understand much.
https://github.com/COVESA/vsomeip/blob/13f9c89ced6ffaeb1faf485152e27e1f40d234cd/documentation/vsomeipUserGuide#L601

lutzbichler, 2021-12-07:
Magic cookies are explained in the SOME/IP specification. They are specific tokens that allow to restart the receiver if a data stream contained data that could not be deserialized as SOME/IP messages.

serdarkose, 2022-05-10:
What happens if the data can't be serialized and magic cookies are not enabled?

Does magic cookies resolve an error state by restarting the receiver?

When is it recommended to use magic cookies or not to use?

Btw, why is it named cookies? I thought about web cookies' functionality. Perhaps more self explaionatory naming would be better.

goncaloalmeida, 2023-05-19:
@serdarkose you should address that questions to the people that write the SOME/IP specification (autosar), not in one project that implements the specification.