COVESA/vsomeip #295: Error : Missing files for android build.

RohitIti88, 2021-11-16:
Error : 
vsomeip/implementation/helper/1.70/boost/asio/basic_socket_ext_local.hpp:22:10: fatal error: 'boost/asio/detail/io_object_impl.hpp' file not found
#include <boost/asio/detail/io_object_impl.hpp>

io_object_impl.hpp and some other files are missing from the folder. Lot of compilation errors. 
mm command always fails to build for android

nkh-lab, 2021-11-16:
Do you have Boost on your AOSP tree? If no here is example how to apply it https://github.com/nkh-lab/aosp-boost.

RohitIti88, 2021-11-17:
Hello COVESA/vsomeip,
Yes I have followed this step and boost is also available in aosp tree.

Cmake .. and make works but mma dosent.

Thanks and regards,
  Rohit iti

RohitIti88, 2021-11-17:
Hello  Nikolay,
Thank you . I followed this step and initially it didnt work. this is because ofthe issue at my AOSP tree.
Thank you for highlighting . It works fine now.

ByteXiaoTang, 2021-12-10:
Sometimes you may compilie under aach64,  cross compilier. You should build boost first

RohitIti88, 2022-08-12:
issue was with local AOSP tree.

closed 2022-08-12 by RohitIti88