COVESA/vsomeip #301: How to configure the overlay so that this allows different network addresses to be managed from a single process?

zhengzhe2, 2021-12-10:


ByteXiaoTang, 2021-12-10:
目前原生是不支持的，即使这样做了也会有冲突。特别是不同stub放在一个进程中。以及配置文件，环境变量设置，需要更改不少代码

zhengzhe2, 2021-12-10:
> 目前原生是不支持的，即使这样做了也会有冲突。特别是不同stub放在一个进程中。以及配置文件，环境变量设置，需要更改不少代码

我配置完之后，发现服务端口的ip地址与“routing”指定的服务所配置的地址一致，导致一个进程中只有一个ip地址生效，若想让其他服务的ip地址生效，只能重新配置“routing”。能帮忙分析下原因吗

ByteXiaoTang, 2021-12-10:
只有后面读的那份配置文件起作用了，覆盖了前面读的配置文件发自我的荣耀手机-------- 原始邮件 --------发件人： breeze ***@***.***>日期： 2021年12月10日周五 14:55收件人： COVESA/vsomeip ***@***.***>抄送： ByteXiaoTang ***@***.***>, Comment ***@***.***>主    题： Re: [COVESA/vsomeip] How to configure the overlay so that this allows different network addresses to be managed from a single process? (Issue #301)

目前原生是不支持的，即使这样做了也会有冲突。特别是不同stub放在一个进程中。以及配置文件，环境变量设置，需要更改不少代码

我配置完之后，发现服务端口的ip地址与“routing”指定的服务所配置的地址一致，导致一个进程中只有一个ip地址生效，若想让其他服务的ip地址生效，只能重新配置“routing”。能帮忙分析下原因吗

zhengzhe2, 2021-12-10:
我的配置文件如下：
vsomeip：
{
    "applications" : [
        {
            "id" : "0x1234",
            "name" : "A",
            "overlay" : "vsomeip_A.json"
        },
        {
           "id" : "0x1235",
           "name" : "B",
           "overlay" : "vsomeip_B.json"
        }
    ],
    "routing" : "A", 
	"seivices" ： [
		{
			// A
		}，
		{
			// B
		}
	]
}
程序运行开始之后，加载的是“vsomeip.json”这个文件，当“A”应用初始化的时候，加载“vsomeip_A.json”配置文件；当“B”应用初始化的时候，加载“vsomeip_B.json”配置文件。
当我设置“routing”为“A”时， “A”服务和“B”服务的端口ip都是“A”；当设置“routing”为B时，“A”服务和“B”服务的端口ip都是“B”。
我的期望是，“A”服务的端口ip是“A”， “B”服务的端口ip是“B”
没太明白 你所说的“只有后面读的那份配置文件起作用了，覆盖了前面读的配置文件”的意思？实际并没有覆盖，只是被设置为“routing”的应用的ip生效而已。

closed 2021-12-14 by zhengzhe2

zhengzhe2, 2021-12-14:
解决方案：
1.配置vsomeip.json文件：
 {					
   "applications" : [					
      {					
         "id" : "0x1234",					
         "name" : "A"					
      },					
      {					
         "id" : "0x5678",					
         "name" : "B",					
         "overlay" : "vsomeip_b.json"					
      }				
   ],					
   "logging" : {					
      "console" : "true",					
      "level" : "debug"					
   },					
   "netmask" : "255.255.0.0",					
   "service-discovery" : {					
      "cyclic_offer_delay" : "1000",					
      "enable" : "true",					
      "initial_delay_max" : "50",					
      "initial_delay_min" : "10",					
      "multicast" : "224.0.0.1",					
      "port" : "30450",					
      "protocol" : "udp",					
      "repetitions_base_delay" : "50",					
      "repetitions_max" : "3",					
      "request_response_delay" : "1500",					
      "ttl" : "3"					
   },					
   "services" : [					
      {					
         "eventgroups" : [					
            {					
               "eventgroup" : "1",					
               "events" : [ "32765", "32766" ]					
            }					
         ],					
         "events" : [					
            {					
               "event" : "32765",					
               "is_field" : "false",					
               "is_reliable" : "true"					
            },					
            {					
               "event" : "32766",					
               "is_field" : "true",					
               "is_reliable" : "true"					
            }					
         ],					
         "instance" : "1",					
         "major_version" : 0,					
         "minor_version" : 1,					
         "reliable" : "30536",					
         "service" : "4146"					
      },					
      {					
         "eventgroups" : [					
            {					
               "eventgroup" : "2",					
               "events" : [ "32771", "32772" ]					
            }					
         ],					
         "events" : [					
            {					
               "event" : "32771",					
               "is_field" : "false",					
               "is_reliable" : "true"					
            },					
            {					
               "event" : "32772",					
               "is_field" : "true",					
               "is_reliable" : "true"					
            }					
         ],					
         "instance" : "2",					
         "major_version" : 0,					
         "minor_version" : 1,					
         "reliable" : "30517",					
         "service" : "4123"					
      }					
   ],					
   "tracing" : {					
      "enable" : true,					
      "sd_enable" : true					
   },					
   "unicast" : "172.45.3.65"					
}
主要修改点：
1.1. 给新增服务“B”添加“overlay”字段
1.2. 删除“routing”字段 （重点）

2.新增vsomeip_b.json文件：
{
    "unicast" : "172.35.3.24",
    "netmask" : "255.255.0.0",
    "network" : "b"
}

3.添加路由：
sudo route add -net 224.0.0.1 netmask 255.255.255.255 dev enp0s8.404
sudo route add -net 224.0.0.1 netmask 255.255.255.255 dev enp0s8.406

注意：
enp0s8.404  ->  172.45.3.65
enp0s8.406  ->  172.35.3.24