COVESA/vsomeip #302: Current CMake usage is incompatible with Boost 1.70 or newer

pauloasherring, 2021-12-10:
Currently, the CMakeLists.txt relies on Boost_LIBRARY_DIR and Boost_INCLUDE_DIR (and maybe others) family of variables.
Effective from Boost 1.70 and later, the default behavior from FindBoost.cmake is rather different.

From FindBoost.cmake:

```
Boost CMake
^^^^^^^^^^^

If Boost was built using the boost-cmake project or from Boost 1.70.0 on
it provides a package configuration file for use with find_package's config mode.
This module looks for the package configuration file called
``BoostConfig.cmake`` or ``boost-config.cmake`` and stores the result in
``CACHE`` entry "Boost_DIR".  If found, the package configuration file is loaded
and this module returns with no further action.  See documentation of
the Boost CMake package configuration for details on what it provides.

Set ``Boost_NO_BOOST_CMAKE`` to ``ON``, to disable the search for boost-cmake.
#]=======================================================================]
```

Possible solutions, tested locally:
- Pass the option during the cmake generation: `cmake -DBoost_NO_BOOST_CMAKE=ON  ..`
- Hardcode on the CMakeLists.txt the option: `set (Boost_NO_BOOST_CMAKE ON)`

4soos, 2022-01-13:
Hi, there, I tested on my machine and it's fine. 
The command just is `cmake ..` and output :
```bash
-- Found Boost: /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.55") found components: system thread filesystem 
-- Using boost version: 107100
```

by the way:

- cmake version 3.22.1
- OS: Manjaro 21.2.1 Qonos

pauloasherring, 2022-01-13:
Well, this is awkward :running_man: 
Just tested it again now and seems to be working. 
Not sure if by the time I had a defective host, I've formatted my PC since then.
Closing, and thanks for the feedback, @4soos

closed 2022-01-13 by pauloasherring