COVESA/vsomeip #305: How to parse the someip packages by wireshark（version3.6，now UDP only）

ziyangfu, 2021-12-28:
Hi
   wireshark offered the filter of someip and someipsd，but I can not find someip protocol packages like following image。just UDP only now。 
[image]

doschropa, 2022-02-10:
In Settings → Protocols → SOME/IP and SOME/IP-SD you have to enter your configured ports to make it work properly

kheaactua, 2022-02-11:
There doesn't seem to be much (any) docs about this.  I've managed to get it working though, decoding function parameters and the likes.

Having [packet-someip.c](https://github.com/wireshark/wireshark/blob/0011bb6a4c628b14f706288f58a056b7378f3c53/epan/dissectors/packet-someip.c#L72-L79) open while your decoding helps get the IDs it's needs for types.  For base types (primitives) you need `SOMEIP_PAYLOAD_PARAMETER_DATA_TYPE_BASE_TYPE`, and for the "ID Reference" you reference the type def in the "Base Type List"

splywaczyk, 2022-02-14:
You can use this dissector:
https://gitlab.com/automotive-projects/eth-ws-someip

I remember that you'd have to modify few things as there is already someip in Wireshark, but after that it works very well.

ziyangfu, 2022-02-15:
> In Settings → Protocols → SOME/IP and SOME/IP-SD you have to enter your configured ports to make it work properly

yes,that's right. Wireshark identifies application layer protocols by port number, but SOME/IP do not exist default port like http.

goncaloalmeida, 2023-02-17:
Closing ticket: Ticket already answered.

closed 2023-02-17 by goncaloalmeida