COVESA/vsomeip #306: E2E: How to handle E2E_OK, E2E_WRONG_CRC, E2E_ERROR

turboaffe, 2022-01-14:
Hello everybody,
I would like to react on an e2e error, specifically the ones mentioned in enum generic_check_status : check_status_t { E2E_OK, E2E_WRONG_CRC, E2E_ERROR} (from https://github.com/COVESA/vsomeip/blob/13f9c89ced6ffaeb1faf485152e27e1f40d234cd/implementation/e2e_protection/include/e2e/profile/profile_interface/profile_interface.hpp)

Before implementing E2E, I was using only CommonAPI::CallStatus which returns an error when calling a .fidl/.fdepl generated method.

But I cannot find a way on how to access the E2E error codes after the call, only available return value seems to be CommonAPI::CallStatus.

I found a mapping from isValidCRC() to the INVALID_VALUE of CommonAPI::CallStatus in the CommonAPI Event.hpp, but somehow I have a hard time understanding if INVALID_VALUE would then represent any of the above mentioned E2E faults, or all of them, or if there is no mapping at all.

Thanks a lot in advance for any help!

sgpires, 2023-10-30:
Hi @turboaffe,

At the moment those E2E status are not exposed on CommonAPI.
The INVALID_VALUE you get on CallStatus can represent both the E2E_WRONG_CRC or the E2E_ERROR, so there is no distinction at the moment.
However, this does seem like a useful feature, and as such it will be implemented in the future.

Thanks,
Tiago Pires

goncaloalmeida, 2023-10-30:
This is not a **defect**, for more requests/discussions, please use the Discussions tab.

closed 2023-10-30 by goncaloalmeida

GHARBIAchref, 2024-09-21:
Hello @goncaloalmeida 
 I want to use end-to-end protection in my application. Is using end-to-end protection done solely through configuration in .json configuration files on both the client and server sides of the services, or does it also require code implementation within my algorithm? And how can I verify if the E2E protection profile 4 is working correctly in Wireshark? 
 thanks