COVESA/vsomeip #307: Will Events/Attributes work when the service discovery is disabled

SaravanaRaja88, 2022-01-19:
Is there any configuration changes from the vsomeip.json file which can make the events/attributes communication work between the client and server when we disable the service discovery ? 

Or please clarify by disabling the service discovery to false from vsomeip json can we have events/attributes communication between clients/server ?

Thanks a lot in advance!!!

lutzbichler, 2022-02-01:
Events/attributes require service discovery to be enabled. Otherwise, notification cannot work as no subscribe messages are exchanged between client and server.

SaravanaRaja88, 2022-02-01:
@lutzbichler Thanks a lot very much for getting back to me.  It would be very nice if you can clarify one more doubt of mine.

Suppose lets say we would try to give the vsomeip daemon where to look for the required services of the servers.  In this case lets say there is no need for vsomeip daemon to send the find requests to find the services.  Lets keep the service discovery enabled as before.  Can we make such a set up is this possible ?

The intension of this question is sometimes on the network for the clients the services are not available. The required proxy's are not offered.   So just to skip this find -> offer,  but then the client and server communicate over network like normal service discovery enabled.  With all events/attributes works.

fcmonteiro, 2023-11-14:
@SaravanaRaja88 currently that's not possible

closed 2023-11-14 by fcmonteiro