COVESA/vsomeip #308: Boost zlib Remote Code Execution Vulnerability

aelfisha, 2022-01-30:
It was announced that  Boost C++ Libraries versions ≤ 1.77.0 have a "Remote Code Execution" vulnerability.
Is there any intention to update to the boost version >= 1.78.0?

gunnarx, 2022-02-03:
@aelfisha Can you provide a link to the particular CVE or other reference to what you are thinking about?

In this search I did not find which one might apply:
https://cve.mitre.org/cgi-bin/cvekey.cgi?keyword=boost 

Also did not find anything immediately in the release notes:
https://www.boost.org/users/history/version_1_78_0.html

> It was announced

Where was it announced, please?

aelfisha, 2022-02-07:
@gunnarx I think it was reported in the below link.
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2016-9840

goncaloalmeida, 2023-03-15:
Fixed on version 3.3.0 (master branch).

closed 2023-03-15 by goncaloalmeida