COVESA/vsomeip #310: E2E: How to configure 'custom profile' to represent E2E Profile 7

turboaffe, 2022-02-09:
Dear vsomeip maintainers, is it possible to configure a 'custom profile' to be used as profile 7?
Thanks!

sgpires, 2023-10-02:
Hi @turboaffe , at the moment it is only possible to use profile 1, profile 4 and a custom profile that uses the 32-bit Ethernet CRC, so it is not possible to configure any other profile to be used as profile 7.

goncaloalmeida, 2023-10-02:
@turboaffe if you want to add this feature feel free to create a PR.

closed 2023-10-02 by goncaloalmeida

GHARBIAchref, 2024-09-21:
Hello @goncaloalmeida 
 I want to use end-to-end protection in my application. Is using end-to-end protection done solely through configuration in .json configuration files on both the client and server sides of the services, or does it also require code implementation within my algorithm? And how can I verify if the E2E protection profile 4 is working correctly in Wireshark? 
 thanks