COVESA/vsomeip #321: ETS #33 malformed_message reply.

justoneanotheruser, 2022-03-16:
ETS testcase #33 is:
"Check that the DUT can handle an echoUINT8ArrayMinSize which is shorter 
than 3 elements of the array."
Then expected reply is MALFORMED_MESSAGE

For AUTOSAR xml files specify what it should do for each RequestID (what is the length and the type), but is there a way to do something similar with vsomeip? as the type is not send with payload. So how can I inform that for given RequestID ETS should reply with error if the payload does not match requirements?

Ginakira, 2022-10-08:
I think it should be done in the vsomeip application instead of SOME/IP protocol layer.

goncaloalmeida, 2023-03-07:
yes agreed, this is someip specification, not vsomeip project.

closed 2023-03-07 by goncaloalmeida