COVESA/vsomeip #327: Multiple routing managers error.

Shangwei1024, 2022-03-24:
I have enabled ENABLE_MULTIPLE_ROUTING_MANAGERS when building.
when i run "env VSOMEIP_CONFIGURATION=./vsomeip01.json VSOMEIP_APPLICATION_NAME=service01 ./subscribe" first and run "env VSOMEIP_CONFIGURATION=./vsomeip02.json VSOMEIP_APPLICATION_NAME=service02 ./subscribe" second in the terminal . When the second program runs, the first program will report an error.But running the second one first and then running the first one works.I am running both programs on the same linux computer. My computer has two network cards.

The log:
[root@ihawk examples]# env VSOMEIP_CONFIGURATION=./vsomeip01.json VSOMEIP_APPLICATION_NAME=service01 ./subscribe
2022-02-24 10:51:13.842122 [info] Parsed vsomeip configuration in 0ms
2022-02-24 10:51:13.842694 [info] Using configuration file: "./vsomeip01.json".
2022-02-24 10:51:13.842839 [info] Initializing vsomeip application "service01".
2022-02-24 10:51:13.843487 [info] Instantiating routing manager [Host].
2022-02-24 10:51:13.843878 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2022-02-24 10:51:13.844361 [info] Service Discovery enabled. Trying to load module.
2022-02-24 10:51:13.845020 [info] Service Discovery module loaded.
2022-02-24 10:51:13.846752 [info] Application(service01, 0001) is initialized (11, 100).
Client settings [protocol=UDP]
2022-02-24 10:51:13.847467 [info] REGISTER EVENT(0001): [3030.0001.8001:is_provider=false]
2022-02-24 10:51:13.847618 [info] REGISTER EVENT(0001): [3030.0001.8004:is_provider=false]
2022-02-24 10:51:13.847759 [info] REGISTER EVENT(0001): [3030.0001.8002:is_provider=false]
2022-02-24 10:51:13.847891 [info] REGISTER EVENT(0001): [3030.0001.8003:is_provider=false]
2022-02-24 10:51:13.848075 [info] SUBSCRIBE(0001): [3030.0001.0001:ffff:1]
2022-02-24 10:51:13.848294 [info] SUBSCRIBE(0001): [3030.0001.0002:ffff:1]
2022-02-24 10:51:13.848490 [info] Starting vsomeip application "service01" (0001) using 2 threads I/O nice 255
2022-02-24 10:51:13.850277 [info] Watchdog is disabled!
2022-02-24 10:51:13.850758 [info] io thread id from application: 0001 (service01) is: 7ffff7f8a8c0 TID: 17698
2022-02-24 10:51:13.850797 [info] io thread id from application: 0001 (service01) is: 7ffff2492700 TID: 17702
2022-02-24 10:51:13.851341 [info] shutdown thread id from application: 0001 (service01) is: 7ffff3494700 TID: 17700
2022-02-24 10:51:13.851027 [info] main dispatch thread id from application: 0001 (service01) is: 7ffff3c95700 TID: 17699
Service [3030.1] is NOT available.
2022-02-24 10:51:13.852679 [info] REQUEST(0001): [3030.0001:255.4294967295]
2022-02-24 10:51:13.853635 [info] Listening at /tmp/vsomeip-1
2022-02-24 10:51:13.859062 [info] vSomeIP 3.1.20.3 | (default)
2022-02-24 10:51:13.859816 [info] Network interface "eth4.3" state changed: up
2022-02-24 10:51:13.860786 [info] Route "224.0.0.0/4 if: eth4.3 gw: n/a" state changed: up
2022-02-24 10:51:13.861228 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992
2022-02-24 10:51:13.861429 [debug] Joining to multicast group 239.127.3.1 from 172.31.3.48
2022-02-24 10:51:13.861595 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 212992
2022-02-24 10:51:13.863373 [info] SOME/IP routing ready.
2022-02-24 10:51:17.144672 [warning] Didn't receive a multicast SD message for 1100ms.
2022-02-24 10:51:17.145332 [debug] Leaving the multicast group 239.127.3.1 from 172.31.3.48
*** Error in `./subscribe': free(): invalid pointer: 0x00007fffec000a08 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x81329)[0x7ffff6952329]
/usr/local/lib/libvsomeip3.so.3(+0x48ecb)[0x7ffff79edecb]
/usr/local/lib/libvsomeip3.so.3(_ZN10vsomeip_v324udp_server_endpoint_impl14leave_unlockedERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x2b1)[0x7ffff7a77421]
/usr/local/lib/libvsomeip3.so.3(_ZN10vsomeip_v324udp_server_endpoint_impl13join_unlockedERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x63)[0x7ffff7a77753]
/usr/local/lib/libvsomeip3.so.3(_ZN10vsomeip_v324udp_server_endpoint_impl4joinERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x3f)[0x7ffff7a777af]
/usr/local/lib/libvsomeip3-sd.so.3(+0x2eebe)[0x7ffff777debe]
/usr/local/lib/libvsomeip3-sd.so.3(+0x37b41)[0x7ffff7786b41]
/usr/local/lib/libvsomeip3.so.3(+0x16ba46)[0x7ffff7b10a46]
/lib64/libstdc++.so.6(+0xb6bd0)[0x7ffff726dbd0]
/lib64/libpthread.so.0(+0x7ea5)[0x7ffff753aea5]
/lib64/libc.so.6(clone+0x6d)[0x7ffff69cfb0d]
======= Memory map: ========
00400000-0040a000 r-xp 00000000 fc:00 1482261                            /root/Desktop/ysw/vsomeip-master/examples/subscribe
0060a000-0060b000 r--p 0000a000 fc:00 1482261                            /root/Desktop/ysw/vsomeip-master/examples/subscribe
0060b000-0060c000 rw-p 0000b000 fc:00 1482261                            /root/Desktop/ysw/vsomeip-master/examples/subscribe
0060c000-0064d000 rw-p 00000000 00:00 0                                  [heap]
7fffe4000000-7fffe4021000 rw-p 00000000 00:00 0 
7fffe4021000-7fffe8000000 ---p 00000000 00:00 0 
7fffe8000000-7fffe8021000 rw-p 00000000 00:00 0 
7fffe8021000-7fffec000000 ---p 00000000 00:00 0 
7fffec000000-7fffec021000 rw-p 00000000 00:00 0 
7fffec021000-7ffff0000000 ---p 00000000 00:00 0 
7ffff1c92000-7ffff1c93000 ---p 00000000 00:00 0 
7ffff1c93000-7ffff2493000 rw-p 00000000 00:00 0 
7ffff2493000-7ffff2494000 ---p 00000000 00:00 0 
7ffff2494000-7ffff2c94000 rw-p 00000000 00:00 0 
7ffff2c94000-7ffff2c95000 ---p 00000000 00:00 0 
7ffff2c95000-7ffff3495000 rw-p 00000000 00:00 0 
7ffff3495000-7ffff3496000 ---p 00000000 00:00 0 
7ffff3496000-7ffff3c96000 rw-p 00000000 00:00 0 
[... 359 lines cut ...]
        {
            "name" : "service02",
            "id" : "0x0002"
        },
        {    
            "name" : "client-sample",
            "id" : "0x1344"
        }
    ],
    "routing" : "service02",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "239.127.3.1",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "50",
        "repetitions_base_delay" : "50",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "1000",
        "request_response_delay" : "1500"
    }
}

journey175, 2022-04-11:
Is that ok if you deploy the two process in two board?

ByteXiaoTang, 2022-07-18:
have the same applications id on two boards??

goncaloalmeida, 2023-03-03:
@Shangwei1024  Does the problem still occur?

closed 2023-03-14 by goncaloalmeida