COVESA/vsomeip #328: Android build error [branch: master]

dterletskiy, 2022-03-24:
I've identified the following build error for AOSP build:

```
error: external/covesa/someip/vsomeip/Android.bp:176:1: module "vsomeip-helloworld-client" variant "android_vendor.31_x86_64": module source path "external/covesa/someip/vsomeip/examples/hello_world/hello_world_client.cpp" does not exist
error: external/covesa/someip/vsomeip/Android.bp:185:1: module "vsomeip-helloworld-service" variant "android_vendor.31_x86_64": module source path "external/covesa/someip/vsomeip/examples/hello_world/hello_world_service.cpp" does not exist
```

It seems to be the consequence of the following commit:
https://github.com/COVESA/vsomeip/commit/18e8b7f98c3ecc9cf3bc7bd6c5c1098ad397d9ac

=> Could you, please, have a look at this issue? It blocks our work with Some/IP.

4soos, 2022-04-19:
Change the code in `vsomeip/Android.bp` :

```
cc_binary {
    name: "vsomeip-helloworld-client",
    defaults: ["vsomeip_example_defaults"],

    srcs: [
        "examples/hello_world/hello_world_client.cpp",
    ],
}

cc_binary {
    name: "vsomeip-helloworld-service",
    defaults: ["vsomeip_example_defaults"],

    srcs: [
        "examples/hello_world/hello_world_service.cpp",
    ],
}
```
To
```
cc_binary {
    name: "vsomeip-helloworld-client",
    defaults: ["vsomeip_example_defaults"],

    srcs: [
        "examples/hello_world/hello_world_client_main.cpp",
    ],
}

cc_binary {
    name: "vsomeip-helloworld-service",
    defaults: ["vsomeip_example_defaults"],

    srcs: [
        "examples/hello_world/hello_world_service_main.cpp",
    ],
}
```

It works for me and I hope it can help you.

dterletskiy, 2022-05-01:
The question is not how to fix it. I did it locally.
The question is I clone automatically several github repos and build then. But it this case build system does not work without additional patching and so on.

But, anyway, thank for reply.

ByteXiaoTang, 2022-07-18:
you can  compile  the project with the demo not compilie

closed 2023-03-03 by goncaloalmeida