COVESA/vsomeip #331: multiple routing managers can not work on one machine, and other machine is ok ?????

journey175, 2022-04-09:
'network'
Network identifier used to support multiple routing managers on one host. This
setting changes the name of the shared memory segment in /dev/shm and the name
of the unix domain sockets in /tmp/. Defaults to vsomeip meaning the shared
memory will be named /dev/shm/vsomeip and the unix domain sockets will be
named /tmp/vsomeip-$CLIENTID

1. as the user guide say, "network" config for supporting multiple routing managers on one host, now my problem is that when I  enable and config multiple routing, other machine can find the service, but local client can  not find the local service.
so how the config the 'network' to solve this bug.
2. as my test tells, send 1M bytes data to other machine need 48ms through routingmanagerd, and use application as route need 22ms。

journey175, 2022-04-11:
both someip application will create /tmp/vsomeip-0 if they have been set as route， so the problem is that the client can find the service of server when they were deploy at one board , client initialize as route as well as server. and if the client and server are deploy at two boards , it works fine....

kheaactua, 2022-04-29:
> Defaults to vsomeip meaning the shared memory will be named `/dev/shm/vsomeip`

Is this still the case?  I haven't found shared memory usage in the code (v3.1.20.3).  Plus it should stand out as it would have to be customized for AOSP.

In https://github.com/COVESA/vsomeip/issues/7 , there's the suggestion that it might be replaced.  The change log mentions shared memory race conditions in v2, but I can't find anything definitively stating whether shared memory is still used in the code - except for in the docs for the `network` key.

journey175, 2022-05-05:
> Is this still the case? I haven't found shared memory usage in the code (v3.1.20.3). Plus it should stand out as it would have to be customized for AOSP.
>
> In #7 , there's the suggestion that it [...]

Yes， the “network” key-value pair means the prefix of the unix domain socket file name in /tmp/vsomeip , and I  haven't found shared memory usage in the code (v3.1.20.3) either.

1. when the commuication is established at the local board，the app proxy connect with the local route with unix domain socket,  but if every app were configured as route by set the "routing":"SOMEIP_APPLICATION_NAME", how can they find the local service,
2. that`s the port I am conflused now. or this mode is not supported by the vsomeip, or just not supported until now
 and will be develop in the soon?
thank you very much for your respond promptly!

kheaactua, 2022-05-05:
> and I haven't found shared memory usage in the code (v3.1.20.3) either.

Yeah, I'm pretty sure it was removed, but a phantom of it still exists in the documentation and CHANGELOG.

> how can they find the local service

You should be able to run multiple routing managers using that `network` key/value pair.  The effect is that instead of the default `/tmp/vsomeip-0` (where `/tmp` is the base path), you'll see `/tmp/<network name>-0`.

So all the local apps start up with their designated `VSOMEIP_CONFIGURATION`, get the `network` name and connect to the routing manager at that socket.

journey175, 2022-05-06:
"You should be able to run multiple routing managers using that network key/value pair. The effect is that instead of the default /tmp/vsomeip-0 (where /tmp is the base path), you'll see /tmp/<network name>-0.

So all the local apps start up with their designated VSOMEIP_CONFIGURATION, get the network name and connect to the routing manager at that socket."

Yes， your opinion is right, but I think the vsomeip design mode is not support local service found when two application were deploy at their own routing managerd according to my expert`s phenomenon。
1. I have enabled multiple routing managerd.
2. I have configured different network with the two application both at one host machine.
3. the two app were connecting to their own routing manager

journey175, 2022-05-06:
@kheaactua 
[image]
[image]

these are my expert log for reference

journey175, 2022-05-13:
[https://github.com/COVESA/vsomeip/blob/master/implementation/runtime/src/application_impl.cpp#L106](url)

ByteXiaoTang, 2022-07-18:
you should have different vlans in the same machine

goncaloalmeida, 2023-02-24:
@journey175 Does the problem still occur?

closed 2023-03-03 by goncaloalmeida

wulijiao, 2023-07-28:
> @kheaactua [image] [image]
> 
> 这些是我的专家日志供参考

Hello, is the problem solved? If so, can you ask the solution?