COVESA/vsomeip #340: Data can not be send, service and instance been deleted

liyisecond339, 2022-05-18:
vsomeip V2.7.0

When APP calls API send() to request service, sometimes there are some prints that vsomeip satck deletes some service/instance, at that time we caught ethernet message by tcpdump tool, found the someip request is not sent actually.

2022-05-14 13:43:10.562524 [error] Cannot find free client port!
2022-05-14 13:43:10.562524 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:11.415870 [error] Cannot find free client port!
2022-05-14 13:43:11.418870 [error] Cannot find free client port!
2022-05-14 13:43:11.420870 [error] Cannot find free client port!
2022-05-14 13:43:11.425870 [error] Cannot find free client port!
2022-05-14 13:43:11.428870 [error] Cannot find free client port!
2022-05-14 13:43:11.511870 [error] Cannot find free client port!
2022-05-14 13:43:11.563870 [error] Cannot find free client port!
2022-05-14 13:43:11.564870 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:11.567870 [error] Cannot find free client port!
2022-05-14 13:43:11.569870 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:11.864070 [error] Cannot find free client port!
2022-05-14 13:43:11.867070 [error] Cannot find free client port!
2022-05-14 13:43:11.874070 [error] Cannot find free client port!
2022-05-14 13:43:11.876070 [error] Cannot find free client port!
2022-05-14 13:43:11.878070 [error] Cannot find free client port!
2022-05-14 13:43:12.577254 [error] Cannot find free client port!
2022-05-14 13:43:12.588254 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:12.599254 [error] Cannot find free client port!
2022-05-14 13:43:12.611254 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:12.728254 [error] Cannot find free client port!
2022-05-14 13:43:12.976462 [info] Application/Client 0799 is registering.
2022-05-14 13:43:12.992462 [info] Client [2001] is connecting to [799] at /tmp/vsomeip-799
2022-05-14 13:43:13.021462 [info] Expired subscription (3012.1.1 from 172.31.3.49:33519(0)
2022-05-14 13:43:13.625659 [error] Cannot find free client port!
2022-05-14 13:43:13.637659 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:13.650659 [error] Cannot find free client port!
2022-05-14 13:43:13.673659 [info] REGISTERED_ACK(0799)
2022-05-14 13:43:13.664659 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:14.212803 [info] REQUEST(0799): [102a.0001:1.0]
2022-05-14 13:43:14.221803 [error] Cannot find free client port!
2022-05-14 13:43:14.680987 [error] Cannot find free client port!
2022-05-14 13:43:14.690987 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:14.701987 [error] Cannot find free client port!
2022-05-14 13:43:14.711987 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:14.934179 [error] Cannot find free client port!
2022-05-14 13:43:14.950179 [error] Cannot find free client port!
2022-05-14 13:43:14.945179 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:14.977179 [info] Expired subscription (3002.1.1 from 172.31.3.2:33503(0)
2022-05-14 13:43:14.993179 [error] Cannot find free client port!
2022-05-14 13:43:14.988179 [info] Expired subscription (3010.1.1 from 172.31.3.2:33571(0)
2022-05-14 13:43:15.020179 [info] Expired subscription (3010.1.2 from 172.31.3.2:33571(0)
2022-05-14 13:43:15.038179 [error] Cannot find free client port!
2022-05-14 13:43:15.031179 [info] Expired subscription (3010.2.1 from 172.31.3.2:49517(0)
2022-05-14 13:43:15.062179 [info] Expired subscription (3010.2.2 from 172.31.3.2:49517(0)
2022-05-14 13:43:15.077179 [error] Cannot find free client port!
2022-05-14 13:43:15.072179 [info] Expired subscription (3012.1.1 from 172.31.3.49:33519(0)
2022-05-14 13:43:15.104179 [info] Expired subscription (3015.1.1 from 172.31.3.49:33522(0)
2022-05-14 13:43:15.119179 [error] Cannot find free client port!
2022-05-14 13:43:15.114179 [info] Expired subscription (3015.1.2 from 172.31.3.49:33522(0)
2022-05-14 13:43:15.147179 [info] Expired subscription (3015.1.3 from 172.31.3.49:33522(0)
2022-05-14 13:43:15.157179 [info] Expired subscription (3016.1.1 from 172.31.3.49:33523(0)
2022-05-14 13:43:15.167179 [info] Expired subscription (3018.1.1 from 172.31.3.49:33525(0)
2022-05-14 13:43:15.182179 [info] Expired subscription (3018.1.1 from 172.31.3.2:33525(0)
2022-05-14 13:43:15.194179 [info] Expired subscription (3019.1.1 from 172.31.3.49:33526(0)
2022-05-14 13:43:15.204179 [info] Expired subscription (301b.1.1 from 172.31.3.49:33528(0)
2022-05-14 13:43:15.214179 [info] Expired subscription (301c.1.1 from 172.31.3.49:33529(0)
2022-05-14 13:43:15.224179 [info] Expired subscription (301e.1.1 from 172.31.3.49:33531(0)
2022-05-14 13:43:15.234179 [info] Expired subscription (3030.1.1 from 172.31.3.2:33549(0)
2022-05-14 13:43:15.245328 [info] Expired subscription (3030.1.2 from 172.31.3.2:33549(0)
2022-05-14 13:43:15.398328 [info] update_routing_info: elapsed=205 : delete service/instance b070/1
2022-05-14 13:43:15.727328 [error] Cannot find free client port!
2022-05-14 13:43:15.738328 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:15.748482 [error] Cannot find free client port!
2022-05-14 13:43:15.758482 [error] Routing error. Client from remote service could not be found!
[... 343 lines cut ...]
2022-05-14 13:43:35.957406 [info] Subscription accepted: eventgroup: 1 : target: 172.31.3.49:33523 unreliable from unknown client.
2022-05-14 13:43:35.959406 [info] on_unsubscribe: target: 172.31.3.49:33525 unreliable
2022-05-14 13:43:35.959406 [info] Subscription accepted: eventgroup: 1 : target: 172.31.3.49:33525 unreliable from unknown client.
2022-05-14 13:43:35.960406 [info] on_unsubscribe: target: 172.31.3.49:33526 unreliable
2022-05-14 13:43:35.960406 [info] Subscription accepted: eventgroup: 1 : target: 172.31.3.49:33526 unreliable from unknown client.
2022-05-14 13:43:35.961406 [info] on_unsubscribe: target: 172.31.3.49:33528 unreliable
2022-05-14 13:43:35.961406 [info] Subscription accepted: eventgroup: 1 : target: 172.31.3.49:33528 unreliable from unknown client.
2022-05-14 13:43:35.961406 [info] on_unsubscribe: target: 172.31.3.49:33529 unreliable
2022-05-14 13:43:35.963406 [info] Subscription accepted: eventgroup: 1 : target: 172.31.3.49:33529 unreliable from unknown client.
2022-05-14 13:43:35.964406 [info] on_unsubscribe: target: 172.31.3.49:33531 reliable
2022-05-14 13:43:35.966406 [info] Subscription accepted: eventgroup: 1 : target: 172.31.3.49:33531 reliable from unknown client.
2022-05-14 13:43:36.057406 [error] Cannot find free client port!
2022-05-14 13:43:36.058406 [error] Cannot find free client port!
2022-05-14 13:43:36.060406 [error] Cannot find free client port!
2022-05-14 13:43:36.061406 [error] Cannot find free client port!
2022-05-14 13:43:36.063406 [error] Cannot find free client port!
2022-05-14 13:43:36.147406 [error] Cannot find free client port!
2022-05-14 13:43:36.149406 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:36.152406 [error] Cannot find free client port!
2022-05-14 13:43:36.156406 [error] Cannot find free client port!
2022-05-14 13:43:36.153406 [error] Routing error. Client from remote service could not be found!
2022-05-14 13:43:36.267589 [error] Cannot find free client port!
2022-05-14 13:43:36.276589 [error] Cannot find free client port!
2022-05-14 13:43:36.277589 [error] Cannot find free client port!
2022-05-14 13:43:36.280589 [error] Cannot find free client port!

ByteXiaoTang, 2022-07-18:
the port may not allocate or has some one bind and you bind error

joao-d, 2023-03-07:
@liyisecond339 Does the problem still occur?

closed 2023-03-14 by goncaloalmeida