COVESA/vsomeip #346: The version in the Find messages might be wrong

kheaactua, 2022-05-27:
When inspecting with Wireshark, my `OFFER` messages contain the proper version, but my `FIND` messages do not.

_e.g._
```
Offer Service Entry (Service ID 0x0602, Instance ID 0x5679, Version: 1.0)
```

```
Find Service Entry (Service ID 0x0602, Instance ID 0x5679, Version: 1.4294967295)
```

It doesn't seem to matter.  The client knows the interface version is actually 1.0, as `proxy_->getInterfaceVersion()`* is a `Version(1,0)`, and the routing manager will still connect these two.  But it's still suspicious.

\* In addition, in the generated code, my `test_service.hpp` has:
```cpp
45:CommonAPI::Version test_service::getInterfaceVersion() {
46:    return CommonAPI::Version(1, 0);
```
and in `test_serviceSomeIPProxy.cpp` I have
```cpp
125:void test_serviceSomeIPProxy::getOwnVersion(uint16_t& ownVersionMajor, uint16_t& ownVersionMinor) const {
126:    ownVersionMajor = 1;
127:    ownVersionMinor = 0;
```
Basically the version is right everywhere in the code.

fcmonteiro, 2023-04-13:
Hi,
The 4294967295 (0xFFFF FFFF) in the minor part of the version means any version shall be returned.

[image]

cc: @goncaloalmeida

kheaactua, 2023-04-13:
Yep, I had found that.  I did a local modification in capicxx-someip-runtime to enforce the minor version.  Thanks