COVESA/vsomeip #348: Port configuration missing for [1111.2222]. Service is internal.

meiGming, 2022-06-13:
Port configuration missing for [1111.2222]. Service is internal.
helloworld-json
{
    "unicast":"172.20.10.3",
    "logging":
    {
        "level":"debug",
        "console":"true"
    },

    "applications":
    [
        {
            "name":"hello_world_service",
            "id":"0x4444"
        },

        {
            "name":"hello_world_client",
            "id":"0x5555"
        }
    ],

    "services":
    [
        {
            "service":"0x1111",
            "instance":"0x2222",
            "unreliable":"30509"
        }
    ],

    "routing":"hello_world_service",
    "service-discovery":
    {
        "enable":"true"
        "port" : "30490",
        "protocol" : "_udp_",
        "initial_delay_min" : "100",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "15",
        "ttl" : "100",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
service and client used same json file,
some one fix this issue??tell me the soution

Thanks.

journey175, 2022-06-24:
look for other issue , someone has fix this question。that is adding {serviceId、instanceId、port} in router`s json file.

ByteXiaoTang, 2022-07-18:
service-discovery no muticast

closed 2023-03-07 by goncaloalmeida