COVESA/vsomeip #350: why notify api always use 2000 us ? weather there have a parameter to control the time delay

zhiwei2018, 2022-06-14:
in notify-sample.cpp ,the code is:
  **while (is_offered_ && running_) {
      if (its_size == sizeof(its_data))
          its_size = 1;

      for (uint32_t i = 0; i < its_size; ++i)
          its_data[i] = static_cast<uint8_t>(i);

      {
          std::lock_guard<std::mutex> its_lock(payload_mutex_);
          payload_->set_data(its_data, its_size);

          std::cout << "Setting event (Length=" << std::dec << its_size << ")." << std::endl;

          ### app_->notify(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, payload_);

      }

      its_size++;

      std::this_thread::sleep_for(std::chrono::milliseconds(cycle_));
  }**
if i set the cycle is >2ms, the api(app_->notify) use time  20us-40us
if i set the cycle is 1ms, the api(app_->notify) use time  1ms
if i set the cycle is 0ms, the api(app_->notify) use time  2ms

how can i let the data send to client in 500us or less ?  weather there have a json parameter or api parameter to control the time delay

ByteXiaoTang, 2022-07-18:
has deque in routing_manager

fengtuo58, 2022-11-19:
config jitters

closed 2023-03-14 by goncaloalmeida