COVESA/vsomeip #352: How to set security protocols e.g. SecOC in vsomeip?

lh315936716, 2022-06-17:
There are sevral security protocols supported by AUTOSAR AP. I didn't find anything related to this in vsomeip manual/code and I didn't see anybody talk about this. I'm wondering if vsomeip could configure security protocl or if this should be done in other part of the whole process. Thank you!

Iandiehard, 2022-09-08:
Normally since vsomeip is the first receiver of pdu's, therefore it should first distinguish whether its a SECOC protected pdu or normal pdu from its configuration. And then it should ask Crypto library for verification. Unfortunately I think its not supported by vsomeip

lh315936716, 2022-09-13:
> 

> Hi landiehard, thank you so much! I figured that SecOC is pretty much similar with traditional verification method like CRC which is also adding some extra bits at the end of the payload. I guess it might be only related to the serialization part and thus has nothing to do with SOMEIP protocol. Do you think it is possiblely valid if I manualy add  the SecOC information to the end of payload？Thank you again. I didn't expect I could get a reply after so long.

goncaloalmeida, 2023-05-26:
This feature is not planned internally. Feel free to do it and create a PR if you want.

closed 2023-05-26 by goncaloalmeida