COVESA/vsomeip #353: can't auto find service between two device

AndyYangjd, 2022-06-17:
The server is running on Raspberry Pi（ubuntu22.04）.
```bash
andy@andy-desktop:~/Desktop/vsomeip-3.1.20.3/build$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:cb:fc:fc brd ff:ff:ff:ff:ff:ff
    inet 10.13.1.113/24 brd 10.13.1.255 scope global noprefixroute eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::ef9:8218:3179:40a6/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether dc:a6:32:cb:fc:fd brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.158/24 brd 192.168.8.255 scope global dynamic noprefixroute wlan0
       valid_lft 85336sec preferred_lft 85336sec
    inet6 240e:878:5:4add:175:ad3a:223f:26bb/64 scope global temporary dynamic 
       valid_lft 7014sec preferred_lft 3414sec
    inet6 240e:878:5:4add:5b8a:2248:5f14:b847/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 7014sec preferred_lft 3414sec
    inet6 fe80::949a:fc33:570:3ef5/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
andy@andy-desktop:~/Desktop/hello_world/build$ route add -nv 224.0.2.2 dev eth0
andy@andy-desktop:~/Desktop/hello_world/build$ export VSOMEIP_CONFIGURATION=/home/andy/Desktop/hello_world/helloworld-local.json
andy@andy-desktop:~/Desktop/hello_world/build$ export VSOMEIP_APPLICATION_NAME=helloworld_service
andy@andy-desktop:~/Desktop/hello_world/build$ ./hello_world_service 
2022-05-17 17:55:02.451799 [info] Parsed vsomeip configuration in 1ms
2022-05-17 17:55:02.454351 [info] Using configuration file: "/home/andy/Desktop/hello_world/helloworld-local.json".
2022-05-17 17:55:02.455076 [info] Configuration module loaded.
2022-05-17 17:55:02.455644 [info] Initializing vsomeip application "hello_world_service".
2022-05-17 17:55:02.458161 [info] Instantiating routing manager [Host].
2022-05-17 17:55:02.459513 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2022-05-17 17:55:02.464021 [info] Service Discovery enabled. Trying to load module.
2022-05-17 17:55:02.475676 [info] Service Discovery module loaded.
2022-05-17 17:55:02.477721 [info] Application(hello_world_service, 4444) is initialized (11, 100).
2022-05-17 17:55:02.478903 [info] Starting vsomeip application "hello_world_service" (4444) using 2 threads I/O nice 255
2022-05-17 17:55:02.483024 [info] main dispatch thread id from application: 4444 (hello_world_service) is: ffff853ff100 TID: 3808
2022-05-17 17:55:02.490457 [info] Watchdog is disabled!
2022-05-17 17:55:02.495442 [info] io thread id from application: 4444 (hello_world_service) is: ffff8643a020 TID: 3806
2022-05-17 17:55:02.494854 [info] shutdown thread id from application: 4444 (hello_world_service) is: ffff84bef100 TID: 3809
2022-05-17 17:55:02.499073 [info] vSomeIP 3.1.20.3 | (default)
2022-05-17 17:55:02.499966 [info] io thread id from application: 4444 (hello_world_service) is: ffff7f7ef100 TID: 3811
2022-05-17 17:55:02.499256 [info] OFFER(4444): [1111.2222:0.0] (true)
2022-05-17 17:55:02.502712 [info] Network interface "eth0" state changed: up
2022-05-17 17:55:02.506201 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992
2022-05-17 17:55:02.512576 [info] Listening at /tmp/vsomeip-4444
2022-05-17 17:55:12.502399 [info] vSomeIP 3.1.20.3 | (default)
2022-05-17 17:55:22.506221 [info] vSomeIP 3.1.20.3 | (default)
2022-05-17 17:55:32.509651 [info] vSomeIP 3.1.20.3 | (default)
2022-05-17 17:55:42.513500 [info] vSomeIP 3.1.20.3 | (default)
^C2022-05-17 17:55:49.319614 [info] STOP OFFER(4444): [1111.2222:0.0] (true)
2022-05-17 17:55:49.326350 [info] Stopping vsomeip application "hello_world_service" (4444).
andy@andy-desktop:~/Desktop/hello_world/build$ cat ../helloworld-local.json 
{
    "unicast":"10.13.1.113",
    "logging":
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },

    "applications":
    [
        {
            "name":"hello_world_service",
[... 102 lines cut ...]
    [
        {
            "name":"hello_world_client",
            "id":"0x5555"
        }
    ],
    "routing":"hello_world_client",
    "service-discovery":
    {
        "enable" : "true",
        "multicast" : "224.0.2.2",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```

From the log in the console, it shows no error.Is there any config error?

journey175, 2022-06-24:
use tcpdump to catch some diagram on multicast ip address 224.0.2.2 adn port 30490, and analyze wheather there are some multicast diagram with wireshark。

AndyYangjd, 2022-06-28:
Ok, thank u. There is some config error and i have solved it.

closed 2022-06-28 by AndyYangjd

RainingJoe, 2022-06-30:
您好，您是改了哪些地方，最后成功的呢？

linh100298, 2022-07-29:
> @RainingJoe IP和端口配置出了问题，先按照官方的文件设置就好了。

Hello, can you please tell me what is the config that you did wrong in your json file or in linux ethernet config? Thank you