COVESA/vsomeip #358: Help needed in clarification of port type in .json file

Ravi-Teja-konda, 2022-06-30:
Hello All,

Can we have a both reliable and unreliable for a service which implements both methods and events

For example :

{
                    "service" : "1234",
                    "instance" : "0x12130",
                    "unreliable" : "31337",
                    "reliable" : { "port" : "4590", "enable-magic-cookies" : "false" }
                },
                {
                    "service" : "5678",
                    "instance" : "0x9876",
                     "unreliable" : "31337",
                    "reliable" : { "port" : "4590", "enable-magic-cookies" : "false" }
                }
}

In the above services it uses communication as both reliable and unreliable 
Is the service is open for both TCP and UDP for methods and events ?

Can client the connect to any port like TCP- 31337 and UDP-4590  for one service ?

ByteXiaoTang, 2022-07-18:
when you sub. you can set sub type

zhaoxin39913, 2022-07-20:
> 

Of course you can have two ports (TCP and UDP) listening on one service. And both ports are available to connect.  Also you can set the reliability type (TCP or UDP) when calling app->send() to specify it.

goncaloalmeida, 2023-02-17:
Closing ticket: Ticket already answered.

> > 
> 
> Of course you can have two ports (TCP and UDP) listening on one service. And both ports are available to connect. Also you can set the reliability type (TCP or UDP) when calling app->send() to specify it.

closed 2023-02-17 by goncaloalmeida