COVESA/vsomeip #361: What's the difference between ET_SELECTIVE_EVENT and ET_EVENT

zhaoxin39913, 2022-07-11:
Currently I am testing vsomeip event APIs of offer_event() and request_event(), and there is a "event_type_e" type input parameter which can be ET_EVENT or ET_SELECTIVE_EVENT or ET_FIELD. I know the differences between ET_FIELD and ET_EVENT, but I don't know the differences between ET_EVENT and ET_SELECTIVE_EVENT and no such sample apps can be found.

Can you provide further clarfications on the difference between ET_SELECTIVE_EVENT and ET_EVENT and the typical user scenario ?

CTWHB, 2023-11-10:
Selective events are not part of the AUTOSAR SOME/IP specification. They are specific to the GENIVI SOME/IP implementation to match the Franca selective broadcasts behavior.
Unlike in the regular broadcasts where the notification will be sent to all connected clients, selective broadcasts may
not be sent to all clients. The server is given the responsibility of selecting which of the clients to send the notification.

closed 2023-11-10 by goncaloalmeida

shivesh1606, 2025-12-24:
Hello @zhaoxin39913  @CTWHB 
I am looking for exact documentation on ET_FIELD and ET_EVENT 
and some refernence how they are working in vsomeip.
Can you guide me to it.