COVESA/vsomeip #362: Vsomeip support on kirkstone #yocto #kirkstone

madhugit2022, 2022-07-20:
I am trying vsomeip on the latest yocto/poky branch kirkstone. I modified the yocto recipe to some extent. Hardcoded Vsomeip to be 3.1.20.1 with no patches and chose BOOST 1.74 after lot of experiments and errors. But still compilation fails. I get a string of errors like /basic_socket_ext_local.hpp:133:7: error: no matching function for call to 'boost::asio::detail::io_object_implboost::asio::detail::reactive_socket_service_ext_local<boost::asio::local::stream_protocol_ext, boost::asio::executor>::io_object_impl(boost::asio::io_context&)'

and so on.

Can anyone tell me the right BOOST version that is compatible with VSOMEIP version? or any patch for 3.1.20 (the patch that came 0001-Support-boost-v1.66.patch didnt work either) 

Any kirkstone recipe for VSomeip shall be appreciated too

bwenm, 2022-08-04:
Have you checked if the correct boost recipe is used? You can do that with bitbake -e boost | grep ^PV.
For me after a downgrade to boost 1.74 Vsomeip is compiling under kirkstone.

madhugit2022, 2022-08-05:
hi bwenm,

Can you please post the bb file ?

madhugit2022, 2022-08-22:
Thanks for the response
my local.conf looks like this
 /home/ubuntu/Madhu/yocto/poky/meta \
  /home/ubuntu/Madhu/yocto/poky/meta-poky \
  /home/ubuntu/Madhu/yocto/poky/meta-yocto-bsp \
  /home/ubuntu/Madhu/yocto/poky/meta-openembedded/meta-oe \
  /home/ubuntu/Madhu/yocto/poky/meta-openembedded/meta-networking \
  /home/ubuntu/Madhu/yocto/poky/meta-openembedded/meta-python \
  /home/ubuntu/Madhu/yocto/poky/meta-openembedded/meta-filesystems \

can you confirm whether i can place this boost in a folder like /home/ubuntu/Madhu/yocto/meta-boost?

bwenm, 2022-08-22:
Yes something like that should work. Create your own layer with BBFILE_PRIORITY = "8" and add it to your local.conf.

madhugit2022, 2022-09-01:
Hi,

Can you please provide the vsomeip recipe too?

MoonGyu-Jeong, 2023-01-05:
@ghost Thanks to your comment, I can bitbake commonapi vsomeip in my simple image.
@madhugit2022 This is my vsomeip recipe version 3.1.20.3

```
SUMMARY = "the implementation of SOME/IP"
SECTION = "base"
LICENSE = "MPLv2"
LIC_FILES_CHKSUM = "file://LICENSE;md5=815ca599c9df247a0c7f619bab123dad"

DEPENDS = "boost dlt-daemon"
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

SRCREV = "13f9c89ced6ffaeb1faf485152e27e1f40d234cd"
SRC_URI = "git://github.com/GENIVI/${BPN}.git;protocol=https \
    "
S = "${WORKDIR}/git"

inherit cmake lib_package gitpkgv

PACKAGES:remove = "${PN}-bin"
FILES:${PN} += "${bindir}/vsomeipd ${sysconfdir}/${BPN}"
FILES:${PN}-dev += "${libdir}/cmake"

BBCLASSEXTEND = "nativesdk"

do_install:append() {
    mv ${D}/usr/etc ${D}
}
```

closed 2023-03-03 by goncaloalmeida

jsb9945, 2023-11-02:
@MoonGyu-Jeong Did you create a recipe using devtool??

MoonGyu-Jeong, 2023-11-03:
> @MoonGyu-Jeong Did you create a recipe using devtool??

No, I didn't use it

jsb9945, 2023-11-03:
@MoonGyu-Jeong Thank you for your answer. I built an image by adding it to the above recipe. Is there a way to check the someip motion?

Shishir-Pokhrel, 2024-02-25:
Could you guys also build the examples directory and use it in the final yocto image? I am trying to do this since two months but it is not working. I have the routingmanagerd shipped in the bin directory as well. The routingmanagerd is inside the examples directory. How can I make sure that the examples are compiled and shipped to the final image?