COVESA/vsomeip #363: Service [1234.5678] is NOT available from client side

dcs2020, 2022-07-21:
Hi,

I am using notify-sample and subscribe-sample from examples in build folder, when I am using same system it is working fine. when I am using two different systems I am getting the error service is not available.

Below are client logs:
2022-06-21 13:57:11.690590 [info] Parsed vsomeip configuration in 1ms
2022-06-21 13:57:11.693530 [info] Using configuration file: "client.json".
2022-06-21 13:57:11.694437 [info] Configuration module loaded.
2022-06-21 13:57:11.695010 [info] Initializing vsomeip application "subscribe-sample".
2022-06-21 13:57:11.696041 [info] Instantiating routing manager [Host].
2022-06-21 13:57:11.698670 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2022-06-21 13:57:11.700118 [info] Service Discovery enabled. Trying to load module.
2022-06-21 13:57:11.716688 [info] Service Discovery module loaded.
2022-06-21 13:57:11.719028 [info] Application(subscribe-sample, 1313) is initialized (11, 100).
Client settings [protocol=UDP]
2022-06-21 13:57:11.720426 [info] REGISTER EVENT(1313): [1234.5678.8778:is_provider=false]
2022-06-21 13:57:11.721162 [info] SUBSCRIBE(1313): [1234.5678.4465:ffff:0]
2022-06-21 13:57:11.721868 [info] notify_one_unlocked: Notifying 8778 to client 1313 failed. Event payload not set!
2022-06-21 13:57:11.722663 [info] Starting vsomeip application "subscribe-sample" (1313) using 2 threads I/O nice 255
2022-06-21 13:57:11.725381 [info] main dispatch thread id from application: 1313 (subscribe-sample) is: b68c93e0 TID: 2030
2022-06-21 13:57:11.726032 [info] shutdown thread id from application: 1313 (subscribe-sample) is: b60c93e0 TID: 2031
2022-06-21 13:57:11.731852 [info] Watchdog is disabled!
Service [1234.5678] is NOT available.
2022-06-21 13:57:11.733700 [info] REQUEST(1313): [1234.5678:255.4294967295]
2022-06-21 13:57:11.734126 [info] io thread id from application: 1313 (subscribe-sample) is: b6f93000 TID: 2029
2022-06-21 13:57:11.735802 [info] Listening at /tmp/vsomeip-1313
2022-06-21 13:57:11.736160 [info] vSomeIP 3.1.20.3 | (default)
2022-06-21 13:57:11.737705 [info] io thread id from application: 1313 (subscribe-sample) is: b4cff3e0 TID: 2033
2022-06-21 13:57:21.737952 [info] vSomeIP 3.1.20.3 | (default)

Server Logs:
2022-06-20 20:41:01.601781 [info] Parsed vsomeip configuration in 1ms
2022-06-20 20:41:01.604784 [info] Using configuration file: "server.json".
2022-06-20 20:41:01.605741 [info] Configuration module loaded.
2022-06-20 20:41:01.606339 [info] Initializing vsomeip application "notify-sample".
2022-06-20 20:41:01.607240 [info] Instantiating routing manager [Host].
2022-06-20 20:41:01.609741 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2022-06-20 20:41:01.611240 [info] Service Discovery enabled. Trying to load module.
2022-06-20 20:41:01.628063 [info] Service Discovery module loaded.
2022-06-20 20:41:01.630307 [info] Application(notify-sample, 1212) is initialized (11, 100).
2022-06-20 20:41:01.631569 [info] REGISTER EVENT(1212): [1234.5678.8778:is_provider=true]
2022-06-20 20:41:01.632470 [info] Starting vsomeip application "notify-sample" (1212) using 2 threads I/O nice 255
2022-06-20 20:41:01.632667 [info] OFFER(1212): [1234.5678:0.0] (true)
2022-06-20 20:41:01.635810 [info] Listening at /tmp/vsomeip-1212
Setting event (Length=1).
2022-06-20 20:41:01.637819 [info] Watchdog is disabled!
2022-06-20 20:41:01.639727 [info] io thread id from application: 1212 (notify-sample) is: b6f42000 TID: 2194
2022-06-20 20:41:01.641152 [info] vSomeIP 3.1.20.3 | (default)
2022-06-20 20:41:01.641545 [info] shutdown thread id from application: 1212 (notify-sample) is: b4dff3e0 TID: 2198
2022-06-20 20:41:01.638546 [info] main dispatch thread id from application: 1212 (notify-sample) is: b55ff3e0 TID: 2197
Application notify-sample is registered.
2022-06-20 20:41:01.645234 [info] io thread id from application: 1212 (notify-sample) is: b3dff3e0 TID: 2200
Setting event (Length=2).
Setting event (Length=3).
Setting event (Length=4).
Setting event (Length=5).
Setting event (Length=6).
^C2022-06-20 20:41:07.696580 [info] STOP OFFER(1212): [1234.5678:0.0] (true)
2022-06-20 20:41:07.646225 [info] Stopping vsomeip application "notify-sample" (1212).

Thank you.

goncaloalmeida, 2023-03-03:
@dcs2020 could you please provide the configurations that you are using?

aadarshkt, 2024-10-30:
Hi @goncaloalmeida, 

I am facing similar issue of client not detecting service, eventhough offer packets are visible at the interface. 

I have created an issue here. [Link to the issue](https://github.com/COVESA/vsomeip/issues/800)