COVESA/vsomeip #364: how to segment and reassemble  message? any example?

UItogether, 2022-07-23:
under the situation of big payload, how to segment and reassemble message?

feuillatte, 2022-08-19:
You will have to create your own serialization and deserialization functions for your payloads. SOME/IP is not a high-level protocol. 

In the simplest case, you can recast a 4-byte payload to a 4-byte variable like an uint64_t. Anything more complex requires you to start incrementing pointer addresses based on the payload byte array traversal and extract out your data objects based on predefined byte-offsets from the start of the byte array.

ims0, 2022-11-24:
configure the SOME/IP-TP feature

closed 2023-03-07 by goncaloalmeida

jafar75, 2023-09-24:
@feuillatte I have a question about ser/deser functions. can we use some tools like cap'n proto or protobuf at this stage?

feuillatte, 2023-09-26:
@jafar75 I haven't personally worked with protobuf or cap'n proto but any serialization-deserialization library should work in principle. SOME/IP only sees the message payload as an array of bytes, and it's up to the user to deal with the serialization and deserialization. In automotive, one typically has some kind of message layout definitions in an Interface Definition Language (IDL) that is then used by a code-generator to produce the ser-des code to and from the byte-representations.

Dealing with packets beyond the single message limit (some 1400 bytes) requires SOME/IP-TP to be used to reassemble the total payload before deserializing it.

AnandShastry, 2025-04-23:
Protobuf can be used as serialization / deserialization technic