COVESA/vsomeip #374: Need help for load data, load service interface or use of service id , instance id from config json file

aparna-ramachandran, 2022-08-24:
I need to use the service id,instance id's of client and server from confiuration json file for establishing communication instead of hardcoded id's.I could not find any resources online but I could see api's load_data,load_service,load_clients in configuration_impl.hpp,(which loads the json file and extracts the relevant id's)however these api's are not provided in interfaces.
Could someone please let me know how to go about this?

Shoolapani, 2022-08-25:
Yes, Please anyone help me also, I am stuck at this. Please show us some example.

ims0, 2022-09-07:
The hardcoded id needs to be present in the code, because different callbacks need to be registered for different service instances, The service instance id in the configuration file should be consistent with the definition in the code.

Reference documentation：https://github.com/COVESA/vsomeip/wiki/vsomeip-in-10-minutes

aparna-ramachandran, 2022-09-07:
@ims0 thanks for confirming

mohitrao12, 2022-11-24:
> I need to use the service id,instance id's of client and server from confiuration json file for establishing communication instead of hardcoded id's.I could not find any resources online but I could [...]

I'm also trying to do that and i figured out that in configuration_imp.cpp we are getting stored values of service id and instance id in "services_" through this map function.  line 427: std::map<service_t, std::map<instance_t, std::shared_ptr<service> > > services_;
I'm trying to extract or print data which is stored in services_ for that i have written one function. i have attach one text file which  in which i have written the changes i made. after all changes code is compiling. but when Im using this command "env VSOMEIP_CONFIGURATION=../../config/vsomeip-udp-service.json VSOMEIP_APPLICATION_NAME=service-sample ./notify-sample"  im not able to print the values which i want . guys could please help me in that. or could you please share your code which is able to print the values from  services_.
[my_implementation.txt](https://github.com/COVESA/vsomeip/files/10081380/my_implementation.txt)

goncaloalmeida, 2023-03-03:
The original problem is solved, @mohitrao12  if you have a problem, please open a new ticket with all the details.

closed 2023-03-03 by goncaloalmeida