COVESA/vsomeip #375: Why is this designed: After the service is restarted, the client is no longer subscribing to the service

ims0, 2022-09-07:
When the client detects that the service has restarted, it will delete the record of the client's subscription service. Then, after the service is restarted, the client will not send the sub information after receiving the offer msg, As a result, service is back to normal, and the client still ignores the service

```
  1031         // Expire all subscriptions / services in case of reboot
  1032         if (is_reboot(_sender, _destination,
  1033                 its_message->get_reboot_flag(), its_message->get_session())) {
  1034             VSOMEIP_INFO << "Reboot detected: IP=" << _sender.to_string();
  1035             remove_remote_offer_type_by_ip(_sender); //
  1036             host_->expire_subscriptions(_sender);         //
  1037             host_->expire_services(_sender);                  //
  1038             if (reboot_notification_handler_) {
  1039                 ip_address_t ip;
  1040                 if (_sender.is_v4()) {
  1041                     ip.address_.v4_ = _sender.to_v4().to_bytes();
  1042                     ip.is_v4_ = true;
  1043                 } else {
  1044                     ip.address_.v6_ = _sender.to_v6().to_bytes();
  1045                     ip.is_v4_ = false;
  1046                 }
  1047                 reboot_notification_handler_(ip);
  1048             }
  1049         }

```

fengtuo58, 2022-10-17:
vsomeip support callback to user,

Eternal6149, 2023-08-15:
Version 3.1.37 seems to have fixed this issue and re subscribed when the peer was launched.
routing_manager_impl -> on_availability

closed 2023-09-01 by fcmonteiro