COVESA/vsomeip #383: [warning] sei::send_cbk received error: Operation not permitted (1) 224.244.224.246:30490 1 56 (0000):

litbubo, 2022-09-26:
[image]
When I try to send a message to the multicast, I get this warning. I tried to capture the packets of all network cards of Linux, but I did not capture some/ip packets. The vsomeip program did not send out service discovery packets. I don't know where the problem is, can anyone help me?

当我尝试向组播中发送报文的时候，就会收到这样的警告。我尝试抓取Linux的所有网卡的包，并没有抓到some/ip的报文，vsomeip的程序确实没有将服务发现报文发送出去。我不知道问题出在哪里，有人能帮帮我吗？
当我尝试向组播中发送报文的时候，就会收到这样的警告。我尝试抓取Linux的所有网卡的包，并没有抓到some/ip的报文，vsomeip的程序确实没有将服务发现报文发送出去。我不知道问题出在哪里，有人能帮帮我吗？

ims0, 2022-11-25:
me

fcmonteiro, 2023-06-05:
Can you check if the problems still occurs with the latest vSomeIP 3.3.8.
Thanks

closed 2023-06-07 by fcmonteiro

quechau, 2024-07-10:
Dear litbubo, fcmonteiro,
Is this issue resolved?
I faced the same issue.
Could you help me? 
Thank you.

manojkumarms1999, 2024-07-10:
Dear @fcmonteiro @litbubo 

faced this issue today on vSomeIP 3.1.20,
the OEM SW is integrated with same version too, so stack upgrade is not a well solution for my case

however, could you explain where the issue occurs and where to check this error:

as per code is says :
 // error: sending of outstanding responses isn't started again
        // delete remaining outstanding responses

Error :
01-01 00:08:48.720  1881  1883 W vsomeip : sei::send_cbk received error: Operation not permitted (1) 233.26.0.5:30490 1 44 (0000): [ffff.8100.0001]

 This error is about a permission related as if I know, I am facing this error on andoird OS, when we changed the system from Permissive to Enforcing the error occurs, I am still adding SELinux permission on android side for solving this error. Probably an AVC denial. lastly even when the system is in permissive mode the error persist.