COVESA/vsomeip #386: Didn't insert subscription as subscription doesn't match reliability type

santhku, 2022-10-17:
Hi, 
The notifier and subscriber code works when i run the notifier code on laptop (linux) and subscriber code on raspberry pi, but  when i am trying to run notify code on raspberry pi and subscriber code on laptop I am getting the below error
 
2022-09-17 13:17:41.156035 [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 1703936
Service [1234.5678] is available.
2022-09-17 13:17:45.139488 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [1234.5678.4465] 2
2022-09-17 13:17:45.143264 [info] update_routing_info: elapsed=1000 : delete service/instance 1234.5678
Service [1234.5678] is NOT available.

Observation is that : The offer packet from the notifier  is on uni-cast ip address (in wireshark)
[image]

How to make the offer service on multicast 

Regard
Santhosh

hari8874, 2022-12-06:
HI Santhku,
I am also facing the same issue,

May i know how you resolved this one.
Kindly help on this

shrinithi98, 2023-09-14:
In request service, add parameter reliability type as rt_unreliable

closed 2023-11-14 by fcmonteiro