COVESA/vsomeip #390: [Question]ipvsd end point

anilbommareddy, 2022-11-18:
As i see in the vsomeip code(latest master branch) there is no option for enabling IPV4sd end point only IPV4 end point option available
is  IPV4sd endpoint obsoleted in vsomeip?
Instead IPv4sd endpoint -can use IPv4 end point option?

gspreeth, 2023-04-12:
I ran into the same problem. I have a stack that offered a service with the SD end point as well as a ipv4 tcp endpoint and since the unknown SD endpoint was first in the list the whole message is discarded after getting a  Unknown remote offer type error.

SOME/IP Service Discovery Protocol
    Flags: 0xc0, Reboot Flag, Unicast Flag
        1... .... = Reboot Flag: Session ID did not roll over since last reboot
        .1.. .... = Unicast Flag: Unicast messages support
        ..0. .... = Explicit Initial Events Flag: Explicit Initial Event control not supported
    Reserved: 0x000000
    Length of Entries Array: 16
    Entries Array
        Offer Service Entry (Service ID 0xa827, Instance ID 0x0001, Version 1.0)
            Type: 0x01 (Offer Service)
            Index 1: 0x01
            Index 2: 0x00
            0001 .... = Number of Opts 1: 0x1
            .... 0000 = Number of Opts 2: 0x0
            Service ID: 0xa827
            Instance ID: 0x0001
            Major Version: 1
            TTL: 6
            Minor Version: 0
    Length of Options Array: 24
    Options Array
        0: IPv4 SD Endpoint Option (198.19.56.203:30490 (UDP))
            Length: 9
            Type: 36
            Reserved: 00
            IPv4 Address: 198.19.56.203
            Reserved 2: 00
            Protocol: 17 (UDP)
            Port: 30490
        1: IPv4 Endpoint Option (198.19.56.203:31003 (TCP))
            Length: 9
            Type: 4
            Reserved: 00
            IPv4 Address: 198.19.56.203
            Reserved 2: 00
            Protocol: 6 (TCP)
            Port: 31003

BrunoLSilvaCTW, 2023-04-12:
@anilbommareddy, @gspreeth, thank you for your feedback! We've also encountered this behavior and will be investigating the issue.

hari8874, 2023-04-26:
Any progress in this issue

if yes please share here

goncaloalmeida, 2023-05-19:
@hari8874 we have the fix done and merged internally, the next vsomeip release will include a fix for this issue.

closed 2023-06-05 by goncaloalmeida

DiogoPedrozza, 2023-06-05:
Fix introduced more precisely to [3.3.7](https://github.com/COVESA/vsomeip/releases/tag/3.3.7):
- "Fix handling of endpoint options"