COVESA/vsomeip #394: [warning] Reliability type for event [5001.4.8777] was not configured Using : RT_UNRELIABLE

mohitrao12, 2022-12-06:
2022-11-06 15:40:36.864209 [warning] Reliability type for event [5001.4.8777] was not configured Using : RT_UNRELIABLE
2022-11-06 15:40:36.864549 [warning] Reliability type for event [5001.4.8778] was not configured Using : RT_UNRELIABLE
2022-11-06 15:40:36.864796 [warning] Reliability type for event [5001.4.8779] was not configured Using : RT_UNRELIABLE

anyone knows how to resolve this warning

yanxiaochun1005, 2022-12-27:
port config is unreliable?

andrea-ortalda, 2023-01-31:
Try to configure the "is_reliable" attribute in the configuration.json file when defining the event:

My example:

2023-00-31 09:52:23.300913 [warning] Reliability type for event [1111.5555.777] was not configured Using : RT_RELIABLE

`{
            "service" : "0x1111",
            "instance" : "0x5555",
            "reliable" : "30500",
            "events" : 
            [
                {
                    "event" : "0x0777",
                    "doorStatus" : "false",
                    "is_reliable" : "true"
                }
            ],
            "eventgroups" :
            [
                {
                    "eventgroup" : "0x6666",
                    "events" : [ "0x777"]
                }
            ]
        },`

closed 2023-03-07 by goncaloalmeida