COVESA/vsomeip #396: Error "**Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)**" means value for the key (routing) in configuration file vsomeip.json:

WWjsnt10, 2022-12-13:
Error "**Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)**" means value for the key (routing) in configuration file vsomeip.json:
... 
"routing" : "**app-name**"
...

and. the argument passed to create_application() call in your application are not matching
...
app = vsomeip::runtime::get()->create_application("**app-name**");
...

_Originally posted by @aananthcn in https://github.com/COVESA/vsomeip/issues/43#issuecomment-605846656_
      

I am try to understand that  vsomeip must have one application name = Configured  "routing" name.   I did some tests, modify the

yanxiaochun1005, 2022-12-27:
host or proxy mode do you want

WWjsnt10, 2022-12-27:
> host or proxy mode do you want
Actually, I am not very clear about the difference between host and proxy mode(**One new routing Manager???**). 

Also, I just want to send the data by UDP via localhost IP in one host machine.  But it is always using the shared memory to do the communication.

yanxiaochun1005, 2022-12-27:
if server and client is same ip in one device ,it is will using the shard memory to do the comminication.

WWjsnt10, 2023-01-31:
> if server and client is same ip in one device ,it is will using the shard memory to do the comminication.

Thanks

closed 2023-01-31 by WWjsnt10