COVESA/vsomeip #398: [question] Why is the configuration module set up as a separate module starting from 2.5.0

w189859254, 2022-12-21:
https://github.com/COVESA/vsomeip/blob/master/CHANGES     :  
v2.5.0
- 。。。
- **Restructured configuration to be a separate module (preparation to enable
  the usage of compiled configurations to speed-up startup)**
- 。。。

_1. First, I don't understand why setting the configuration module as a separate module would improve the startup speed of the application。Until I saw the instructions in vsomeipUserGuide_：

https://github.com/COVESA/vsomeip/blob/master/documentation/vsomeipUserGuide      :
`VSOMEIP_MANDATORY_CONFIGURATION_FILES`: vsomeip allows to specify mandatory configuration
   files to speed-up application startup. While mandatory configuration files are read by all
   applications, all other configuration files are only read by the application that is 
   responsible for connections to external devices. If this configuration variable is not set,
   the default mandatory files vsomeip_std.json, vsomeip_app.json and vsomeip_plc.json are used.

_Second, as I understand it, do you want to reduce the parsing of the configuration file by sharing the configuration, thereby increasing the startup speed of the application? But，according to my tests, parsing the configuration file takes very little time( as shown below - 0ms), so why make the configuration module a separate module?  Is it my misunderstanding? Is there any other consideration?_

configuration_impl.cpp      ：
    std::chrono::steady_clock::time_point end = std::chrono::steady_clock::now();
    VSOMEIP_INFO << "Parsed vsomeip configuration in "
            << std::chrono::duration_cast<std::chrono::milliseconds>(end - begin).count()
            << "ms";

root@ubuntu:/home/pawn/workspace/vsomeip/src/examples# ./notify_server
2022-10-14 02:01:45.204718 [info] Parsed vsomeip configuration **in 0ms**
2022-10-14 02:01:45.204772 [info] Using configuration file: "./vsomeip.json".
2022-10-14 02:01:45.204782 [info] ip = 192.168.0.231
2022-10-14 02:01:45.204802 [info] Default configuration module loaded.
2022-10-14 02:01:45.204808 [info] Initializing vsomeip application "service-sample

lutzbichler, 2023-02-06:
The separation supports speeding up the configuration loading if the module that loads the configuration from a set of json files is replaced by another module that already contains the loaded values ("compiled configuration"). The load time and especially its influence on general system startup heavily depends on the size of the configuration files and the number of applications that load them.

w189859254, 2023-02-17:
[image]

Yeah, I know what you mean. **Is the idea not fully realized yet?**

The image above is from v3.16.1
Line 133，configurion module(contains the loaded values), will not be loaded.
Line 158，in configuration_->load() function, still parse the configuration file.

So, the idea of separating the configuration modules to speed up the startup of the application hasn't worked out yet? Or am I reading it wrong? (ᇂ _ ᇂ | | |)

duartenfonseca, 2024-10-17:
@w189859254 this functionality has a TODO as comment, so that means it is still not fully realized

closed 2024-10-17 by duartenfonseca