COVESA/vsomeip #399: subscribe is not happening

hari8874, 2022-12-22:
trying to connect two hardware boards client(R-car) using Vsomeip(V3.1.20) and server (Infinium tc399 board)someip standards using CAN protocol
Subscribe is not going to happen but able to see the offer messages from server and find messages from client in wireshark.
[image]

Zhaoxiaoliang23, 2023-03-03:
I also met the same problem. How did you solve it?

fcmonteiro, 2023-04-13:
Hi,
Are you still having this issue?
Can you provide us your config files for both machines?
Thanks

cc: @goncaloalmeida

Zhaoxiaoliang23, 2023-04-14:
This problem has been solved, because the VLAN configuration is incorrect. Thanks~

赵晓亮/Xiaoliang Zhao
=========================================================================
经纬恒润 汽车电子事业部 AE诊断工具部
北京市朝阳区酒仙桥路兆维工业园C3座4层，100191 
=========================================================================
Tel: +86 17600805827
Fax: +86 10 64853661
E-mail: ***@***.***

 
From: fcmonteiro
Date: 2023-04-13 17:27
To: COVESA/vsomeip
CC: Moon23; Comment
Subject: Re: [COVESA/vsomeip] subscribe is not happening (Issue #399)
Hi,
Are you still having this issue?
Can you provide us your config files for both machines?
Thanks
cc: @goncaloalmeida

closed 2023-04-14 by goncaloalmeida

hari8874, 2023-04-24:
Hi @Zhaoxiaoliang23 can you tell send the VLAN configuration