COVESA/vsomeip #401: You must communicate using UDP/TCP and you must not use selective events. Local communication does not work. Please note that vSomeIP 3.x has a compatibility layer that enables it usage for application compiled to vSomeIP 2.x.

Arindom-2712, 2023-01-04:
You must communicate using UDP/TCP and you must not use selective events. Local communication does not work. Please note that vSomeIP 3.x has a compatibility layer that enables it usage for application compiled to vSomeIP 2.x.

_Originally posted by @lutzbichler in https://github.com/COVESA/vsomeip/issues/158#issuecomment-724635781_

Arindom-2712, 2023-01-04:
I am also facing a similar issue. (somep2 <-> someip3 communication not established). 

My client is based on someip3 , running on Android.
Server is someip2 based running on linux. 

Using TCP/UDP for communication.

Can you throw some light on the "compatibility layer"  that you talked in your comment. Any special way  to make sure the "compatibility layer" is included in my someip3 based client application.

lutzbichler, 2023-02-06:
Regarding selective events, the compatibility layer will not help. It is only meant to provide the vsomeip 2.x API while using vsomeip3.

closed 2023-03-03 by goncaloalmeida