COVESA/vsomeip #409: Subscription not sent from the client for the offer received [local unreliable port is zero]

vramasubbu, 2023-02-20:
On client start-up. The client is not sending the subscription for the offer message it received.
When we took the some-ip log, the below local unreliable port prints are coming.
When this prints will come? 

[2023-02-15 18:53:48.042] Jan 01 00:00:20.637    someip_client_app.356376..0                 slog      0  insert_subscription: Didn't insert subscription as  local unreliable port is zero: [0005.0001.0001]
[2023-02-15 18:53:48.058] Jan 01 00:00:22.638    someip_client_app.356376..0                 slog      0  insert_subscription: Didn't insert subscription as  local unreliable port is zero: [0001.0001.0001]
[2023-02-15 18:53:48.073] Jan 01 00:00:22.639    someip_client_app.356376..0                 slog      0  insert_subscription: Didn't insert subscription as  local unreliable port is zero: [0005.0001.0001]
[2023-02-15 18:53:48.089] Jan 01 00:00:24.210    someip_client_app.356376..0                 slog      0  vSomeIP 2.10.21 | (default)
[2023-02-15 18:53:48.105] Jan 01 00:00:24.644    someip_client_app.356376..0                 slog      0  insert_subscription: Didn't insert subscription as  local unreliable port is zero: [0001.0001.0001]
[2023-02-15 18:53:48.105] Jan 01 00:00:24.644    someip_client_app.356376..0                 slog      0  insert_subscription: Didn't insert subscription as  local unreliable port is zero: [0005.0001.0001]
[2023-02-15 18:53:48.121] Jan 01 00:00:26.652    someip_client_app.356376..0                 slog      0  insert_subscription: Didn't insert subscription as  local unreliable port is zero: [0001.0001.0001]
[2023-02-15 18:53:48.138] Jan 01 00:00:26.652    someip_client_app.356376..0                 slog      0  insert_subscription: Didn't insert subscription as  local unreliable port is zero: 

Also we noticed, Client is not sending the subscription for the first offer it received.

Issue we are seeing only on the below scenario:
On client startup after first offer it received from the server . as per the application logic Client sends the Fire and Forget method on top of the same service. For that method server sent ICMP port unreachable error message . after that this subscription issues are coming.

Whether this icmp error message from server will leads to this error?

Please share your inputs
it will be helpful.
Thanks.

fcmonteiro, 2023-05-15:
@vramasubbu can you try with the lastest vsomeip version?
We have dropped support to vSomeIP 2.x some time ago.
Thanks

closed 2023-05-19 by goncaloalmeida

Sphinxes0o0, 2023-06-27:
Hi, there, I encountered the same issue on Android 9 (kernel version 4.14) using version 3.1.14.1. Below is the log.

```
vSomeIP : create_eventgroup_entry: Didn't insert subscription as  local unreliable port is zero: [0a01.0001.0001]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as  local unreliable port is zero: [0201.0001.0001]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as  local unreliable port is zero: [0c02.0001.0001]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as  local unreliable port is zero: [0a01.0001.0004]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as  local unreliable port is zero: [0202.0001.0001]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as local reliable port is zero: [0204.0001.0001]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as local reliable port is zero: [0204.0001.0002]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as local reliable port is zero: [0203.0001.0002]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as local reliable port is zero: [0401.0001.0001]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as local reliable port is zero: [0203.0001.0003]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as local reliable port is zero: [0203.0001.0001]
vSomeIP : create_eventgroup_entry: Didn't insert subscription as local reliable port is zero: [0401.0001.0002]
```

Do you have any suggestions ?
Thanks!

goncaloalmeida, 2023-06-27:
@Sphinxes0o0 please test with the last version. We have dropped support to vSomeIP 2.x some time ago.