COVESA/vsomeip #412: server endpoint cannot be created with error saying unavailable port (reliable=unreliable)

hanmhs9, 2023-02-24:
When running applications in docker container inside WSL2, the traditional configuration doesn't work and it fails creating server endpoint. 
The error message says 

[info] vSomeIP 3.1.20.3 | (default)
2023-01-24 09:18:41.552202 [info] Network interface "eth0" state changed: up
2023-01-24 09:18:41.563975 [info] Route "default route (0.0.0.0/0) if: eth0 gw: 192.168.100.1" state changed: up
2023-01-24 09:18:41.636576 [error] create_server_endpoint Server endpoint creation failed. Reason: outbound interface option IPv4: Protocol not available Port: 30490 (reliable=unreliable)
2023-01-24 09:18:41.652631 [error] Service Discovery endpoint could not be created. Please check your network configuration.
2023-01-24 09:18:41.653972 [error] Couldn't start service discovery
2023-01-24 09:18:41.664018 [error] create_server_endpoint Server endpoint creation failed. Reason: outbound interface option IPv4: Protocol not available Port: 30509 (reliable=unreliable)
2023-01-24 09:18:41.665822 [info] SOME/IP routing ready.
2023-01-24 09:18:41.671459 [warning] Route "224.224.224.245/32 if: eth0 gw: n/a" state changed: up
2023-01-24 09:18:51.524091 [info] vSomeIP 3.1.20.3 | (default)
^C2023-01-24 09:18:52.722858 [info] Stopping vsomeip application "World" (1212).
2023-01-24 09:18:52.856743 [info] Exiting vsomeip application...

I suspect the configuration of routing is the reason since it's docker container, but cannot really figure out why and how

hanmhs9, 2023-03-02:
> if it possible to share your json?
> i have no this problem in docker ENV.

Im on a vacation right now. Will get back to u within two days

hanmhs9, 2023-03-03:
Hi,

thanks for the email.

My current environment information is as follows:

WSL2 on Windows

*Windows network*
*ipconfig /all:*
ethernet ipv4: 192.168.0.3  (DNS: 111.118.0.1)
vethernet ipv4 (WSL) : 172.31.0.1

*WSL network*
*ifconfig:*
eth0: 172.31.8.75
lo: 127.0.0.1

Docker container network
ifconfig:
eth0: 172.17.0.2 ~
lo : 127.0.0.1

service.json:
{
    "unicast" : "172.17.0.3",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "World",
            "id" : "0x1212"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509"
        }
    ],
    "routing" : "World",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

client.json
{
    "unicast" : "172.17.0.4",
    "logging" :
    {
[... 104 lines cut ...]
1313 (Hello) is: 5502fd61c0 TID: 41
2023-02-03 11:16:34.568904 [info] Watchdog is disabled!
CLIENT: Service [1234.5678] is NOT available.
2023-02-03 11:16:34.670889 [info] io thread id from application: 1313
(Hello) is: 55147d91c0 TID: 44
2023-02-03 11:16:34.706118 [info] io thread id from application: 1313
(Hello) is: 5501f5b010 TID: 37
2023-02-03 11:16:34.842344 [info] vSomeIP 3.1.20.3 | (default)
2023-02-03 11:16:34.900424 [info] Network interface "eth0" state changed: up
2023-02-03 11:16:34.916348 [info] Route "default route (0.0.0.0/0) if: eth0
gw: 172.17.0.1" state changed: up
2023-02-03 11:16:34.998194 [error] create_server_endpoint Server endpoint
creation failed. Reason: outbound interface option IPv4: Protocol not
available Port: 30490 (reliable=unreliable)
2023-02-03 11:16:34.011855 [error] Service Discovery endpoint could not be
created. Please check your network configuration.
2023-02-03 11:16:34.013217 [error] Couldn't start service discovery
2023-02-03 11:16:34.014938 [info] SOME/IP routing ready.
2023-02-03 11:16:44.860610 [info] vSomeIP 3.1.20.3 | (default)
2023-02-03 11:16:54.866005 [info] vSomeIP 3.1.20.3 | (default)
^C2023-02-03 11:17:04.945611 [info] Stopping vsomeip application "Hello"
(1313).
2023-02-03 11:17:04.058399 [info] Exiting vsomeip application...

Any clues??

magic-tab-01, 2023-03-06:
have you test ping between 172.17.0.3 and 172.17.0.4?
have you check route?
where is 172.17.0.3 and 172.17.0.4?

hanmhs9, 2023-03-06:
Ping between containers works.

What do you mean routes?

172.17.0.3 and 172.17.0.4 are ipv4 of each container at the interface eth0.

They are connected to virtual ethernet address like veth1234 and these are
then connected to docker0 (172.17.0.1) which goes to 0.0.0.0

This 0.0.0.0 is only routed to WSL2’s eth0 interface, which is the route to
the outside internet, via Windows virtual ethernet (vethernet) something
like below (but i didnt touch anything regarding dns inside the docker.
This is what I may need to search)

hanmhs9, 2023-03-07:
I realized that I created the docker image with arm64v8:ubuntu:20.04.

I created the same docker image but from just ubuntu:20.04 and now they
work.

Looks like i have to compile vsomeip application differently in arm64
environment

hanmhs9, 2023-03-07:
Turned out to be different architecture problem (amd64 based application on arm64 environment)

closed 2023-03-07 by hanmhs9

yl2886, 2024-03-22:
bro，same issue with arm64 version running in qemu docker on amd64 environment, have u found the solution?

hanmhs9, 2024-03-22:
Hi, if you are talking about the vsomeip as well,  then I can tell you that
I solved it by using the static routing (not sure if this was the actual
name) instead of the service discovery (which uses the multicast) that
caused the endpoint problem.