COVESA/vsomeip #418: FIND SERVICE sent during Main Phase

manojkumarms1999, 2023-03-13:
Hello everyone, presenting a issue "FIND SERVICE are sent during Main Phase in SOME/IP-SD" that has occured while stress testing 

As we know according to [PRS_SOMEIPSD_00408] Sending Find entries shall be stopped after receiving the corresponding Offer entries by jumping to the Main Phase in which no Find entries are sent.

From the below log we can observe particular service has entered the Main Phase with cyclic_offer_delay of 2 seconds

[image]

on logcat logs particular to the above timing no suspected log observed.

Environment :  on Multiple sleep wakeup scenario between 2 ECU's which uses same SOMEIP Stack 3.1.20.3 
Occurence of the issue : RARE ( 5 /100 ) 

let us know this kind of behaviour is usual / rare or can be ignored too. 👍 

Refered docs :

https://www.autosar.org/fileadmin/standards/foundation/1-4/AUTOSAR_PRS_SOMEIPProtocol.pdf
https://www.autosar.org/fileadmin/standards/classic/22-11/AUTOSAR_SWS_ServiceDiscovery.pdf

manojkumarms1999, 2023-03-21:
Please, let us know this kind of behaviour is usual / rare or can be ignored too.

k08eYLx, 2023-04-01:
can you attach your full pcap file?

manojkumarms1999, 2023-04-19:
[find after answer.zip](https://github.com/COVESA/vsomeip/files/11271656/find.after.answer.zip)
@k08eYLx attached the pcap

manojkumarms1999, 2023-05-11:
@k08eYLx attached the pcap

closed 2023-05-22 by goncaloalmeida