COVESA/vsomeip #424: Different code optimizations, gcc-11 warnings, ostream optimizations

fietzero, 2023-03-21:
I parsed the code and found the following suboptimal solutions:

- std::hex/std::dec and std::setfill() is repeated in many ostream output operations, which is not needed, because those manipulators are persistent (while std::setw() is not)
- many range based for loops iterate by (const) value instead of const reference, this is a waste of code size and CPU, and even gives warnings when compiling with e.g. gcc-11
- std::map operations first do a std::map::find(), and when found something call std::map::operator[](), which searches the map once more, instead of using the iterator returned by find()
- other small issues

Those issues are not really bugs, but they increase the code size and runtime. In embedded systems it's always a good idea not to do that without need.

I already have a local branch with many of those issues being fixed. I can offer to prepare a branch/PR introducing those fixes based on the current master.

fcmonteiro, 2023-04-13:
Hi,
Yes, feel free to do pull requests for your changes.
If they are very different, we would prefer multiple/individual PRs to make it easier for us to analyse and test them.
Thanks

cc @goncaloalmeida

fietzero, 2023-04-27:
I've worked out a branch and checked it locally. How can I push a branch (naming?) to create a PR?

fietzero, 2023-05-04:
Hello Gonçalo,

Double checked, that I have access:

tmp $ git clone --reference ~/src/vsomeip git@github.com:COVESA/vsomeip.git
Cloning into 'vsomeip'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 6 (delta 4), reused 5 (delta 4), pack-reused 0
Receiving objects: 100% (6/6), done.
Resolving deltas: 100% (4/4), completed with 4 local objects.

Now the push:

tmp $ cd ~/src/vsomeip/
src/vsomeip [fix-container-optimizations *$] $ grep url .git/config 
        url = git@github.com:COVESA/vsomeip.git
src/vsomeip [fix-container-optimizations *$] $ git push origin fix-container-optimizations
ERROR: Permission to COVESA/vsomeip.git denied to fietzero.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

Roman

goncaloalmeida, 2023-05-10:
@fietzero thanks for the information, we will check the permissions.

bit2shift, 2023-05-10:
@fietzero push fails because you don't have "push" permission on COVESA repos.

You need to fork the repo and then replace the "origin" URL in your local clone so that you can push to your own fork of vsomeip.

Only after that, you'll be able to open a PR to contribute your proposed changes 😊

fietzero, 2023-05-10:
Hi,

Sorry for the additional workload…

  https://github.com/COVESA/vsomeip/pull/444

Roman

goncaloalmeida, 2023-05-19:
@bit2shift provided the right answer. closing the ticket

closed 2023-05-19 by goncaloalmeida