COVESA/vsomeip #426: Communication between two remote hosts

randheerrk, 2023-03-30:
Hi there,

We are trying to run service client applications on two AWS EC2 instances for a POC. I have verified that both instances are under same VPC and subnet. But When I run the application they are not communicating with each other. Please find the config files I used for the setup below.
client.json
`{
    "unicast" : "client machine ip",
    "netmask" : "255.255.240.0",

    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "true", "path" : "/var/log/vsomeip.log" },
        "dlt" : "true"
    },

    "tracing" :
    {
        "enable" : "true"
    },

    "applications": [
        {
            "name": "app_someip_client",

            "id": "0x5555"
        }
    ],

    "clients" :
    [
        {
            "service" : "0x1111",
            "instance" : "0x2222",
            "unreliable" : "30509"
        }
    ],

    "routing": "app_someip_client",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.244.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}`
server.json
`{
    "unicast":"server machine ip",
    "netmask" : "255.255.240.0",
    "logging":
    {
        "level":"debug",
        "console":"true",
	"file" : { "enable" : "true", "path" : "/var/log/vsomeip.log" },
        "dlt" : "true"
    },

    "tracing" :
    {
        "enable" : "true"
    },

    "applications":
    [
        {
            "name":"app_someip_service",
            "id":"0x4444"
        }
    ],

    "services":
    [
        {
            "service":"0x1111",
            "instance":"0x2222",
            "unreliable":"30509"
        }
    ],

    "routing":"app_someip_service",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.244.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}`

Im not able to fix this issue, Any help on this is much appreciated.

k08eYLx, 2023-04-01:
it maybe not an issue of your json configuration file, pls check your network setup firstly, use tools like tcpdump to make sure the sd offer multicast packets can be captured on both of the two machines.

randheerrk, 2023-04-05:
Thanks for the reply @k08eYLx , turn out it was as issue with setup. I was using EC2 instances for server and client. But by default AWS VPC does not support multicasting. I needed to setup Transit gateway multicasting domain. Once I did that everything works.

I have one more question, So is multicast a must requirement to run server and client in two different hosts? is there any other way we can run it without multicasting

joao-d, 2023-04-14:
In vSomeIP, which is an implementation of the Autosar standard, multicast is used for service discovery. This means that a client can discover the available services in the network by sending a multicast message, and the service providers respond with their service offerings. Multicast is used because it allows the server to advertise its services to multiple clients simultaneously with a single message. This can reduce network traffic and improve efficiency.

k08eYLx, 2023-04-19:
@randheerrk for method communication, vsomeip support "static" service-discovery like the following picture:
1. set "service-discovery" enable to "false";
2. configure service ECU's unicast ip address and port under the "services" section in client ECU's json;
[image]

But it seems event notification communication is not supported by "static" service-discovery, I'm not sure.
And “static” service-discovery is inflexible, it losses the advantage of service-discovery, maybe its usage should be limited.

randheerrk, 2023-04-26:
thanks @k08eYLx , @joao-d

closed 2023-04-26 by randheerrk