COVESA/vsomeip #427: communication between 2 someip machines

mohamedkh147, 2023-03-31:
the client json file :
{
	"unicast" : "192.168.1.176",
		"netmask" : "255.255.255.0",
		"logging" :
		{
			"level" : "debug",
			"console" : "true",
			"file" : { "enable" : "false", "path" : "vsomeip.log" },
			"dlt" : "false"
		},
		"applications" :
			[
			{
				"name" : "client-sample",
				"id" : "0x1000"
			}
			],

				"routing" : "client-sample",
				"service-discovery" :
				{
					"enable" : "true",
					"multicast" : "224.244.224.245",
					"port" : "30490",
					"protocol" : "udp"
				}

}

the sever json 
{
	"unicast" : "192.168.1.178",
		"logging" :
		{
			"level" : "trace",
			"console" : "true",
			"file" : { "enable" : "false" },
			"dlt" : "false"
		},
		"applications" : 
			[
			{
				"name" : "service-sample",
				"id" : "0x1000"
			}
			],
				"services" :
					[
					{
						"service" : "0x10",
						"instance" : "22136",
						"reliable" : { "port" : "30509", "enable-magic-cookies" : "false" },
						"unreliable": "31000"
					}
					],
						"routing" : "service-sample",
						"service-discovery" :
						{
							"enable" : "true",
							"multicast" : "224.244.224.245",
							"port" : "30490",
							"protocol" : "udp"
						}
}

client.sh 
#!/usr/bin/env bash

 sudo route add -net 224.244.224.245 dev enp0s31f6
sudo route add -net 224.0.0.0/4 dev enp0s31f6
 nc -vu 192.168.1.169 30490
COMMONAPI_PATH=../../capicxx-core-runtime/build
COMMONAPI_SOMEIP_PATH=../../capicxx-someip-runtime/build
VSOMEIP_PATH=../../vsomeip/build
export LD_LIBRARY_PATH=../lib:${COMMONAPI_PATH}:${COMMONAPI_SOMEIP_PATH}:${VSOMEIP_PATH}:$LD_LIBRARY_PATH
export VSOMEIP_APPLICATION_NAME=client-sample
export VSOMEIP_CONFIGURATION=../config/client-sample.json
../build/client

server.sh 
 sudo route add -net 224.244.224.245 dev enp0s31f6
sudo route add -net 224.0.0.0/4 dev enp0s31f6
 nc -vu 192.168.1.169 30490
COMMONAPI_PATH=../../capicxx-core-runtime/build
COMMONAPI_SOMEIP_PATH=../../capicxx-someip-runtime/build
VSOMEIP_PATH=../../vsomeip/build
export LD_LIBRARY_PATH=../lib:${COMMONAPI_PATH}:${COMMONAPI_SOMEIP_PATH}:${VSOMEIP_PATH}:$LD_LIBRARY_PATH
export VSOMEIP_APPLICATION_NAME=client-sample
export VSOMEIP_CONFIGURATION=../config/server-sample.json
../build/service

yangqiao95, 2023-05-06:
this json  is  right？

goncaloalmeida, 2023-05-19:
@mohamedkh147 I am not understanding the question/problem on this issue. Could you please provide more info?

closed 2023-06-05 by fcmonteiro