COVESA/vsomeip #428: Systematic Registration issue after link reboot detection

haithem0501, 2023-04-03:
03-31 19:16:25.181  1635  1652 I         : VSIP: Reboot detected: IP=192.168.61.3
03-31 19:16:25.181  1635  1652 I         : VSIP: expire_services for address: 192.168.61.3 : delete service/instance 001e.0001 port [0,0] reliability=false
03-31 19:16:25.181  1635  1652 I         : VSIP: expire_services for address: 192.168.61.3 : delete service/instance 0021.0001 port [0,0] reliability=false
03-31 19:16:25.182  1635  2657 I         : VSIP: ON_UNAVAILABLE(3d1e): [001e.0001:1.1]
03-31 19:16:25.182  1635  1652 I         : VSIP: expire_services for address: 192.168.61.3 : delete service/instance 0023.0001 port [0,0] reliability=false
03-31 19:16:25.182  1635  2655 I         : VSIP: ON_UNAVAILABLE(3d21): [0021.0001:1.0]
03-31 19:16:25.182  1635  1652 I         : VSIP: expire_services for address: 192.168.61.3 : delete service/instance 0024.0001 port [0,0] reliability=false
03-31 19:16:25.182  1635  2828 I         : VSIP: ON_UNAVAILABLE(3d23): [0023.0001:1.0]
03-31 19:16:25.182  1635  1652 I         : VSIP: expire_services for address: 192.168.61.3 : delete service/instance 0025.0001 port [0,0] reliability=false
03-31 19:16:25.182  1635  2684 I         : VSIP: ON_UNAVAILABLE(3d24): [0024.0001:1.0]
03-31 19:16:25.182  1635  1652 I         : VSIP: expire_services for address: 192.168.61.3 : delete service/instance 0027.0001 port [0,0] reliability=false
03-31 19:16:25.182  1635  2835 I         : VSIP: ON_UNAVAILABLE(3d25): [0025.0001:1.0]
03-31 19:16:25.182  1635  1652 I         : VSIP: expire_services for address: 192.168.61.3 : delete service/instance 0028.0001 port [0,0] reliability=false
03-31 19:16:25.182  1635  2677 I         : VSIP: ON_UNAVAILABLE(3d27): [0027.0001:1.0]
03-31 19:16:25.183  1635  1652 I         : VSIP: expire_services for address: 192.168.61.3 : delete service/instance 007f.0001 port [0,0] reliability=false
03-31 19:16:25.183  1635  2849 I         : VSIP: ON_UNAVAILABLE(3d7f): [007f.0001:1.0]
03-31 19:16:25.183  1635  2850 I         : VSIP: ON_UNAVAILABLE(3d28): [0028.0001:1.1]
**03-31 19:16:25.504  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 30 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.504  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 33 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.504  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 35 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.504  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 36 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.504  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 37 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.504  1635  1640 I         : VSIP: configuration_impl:find_specific_port #1: service: 27 instance: 1 reliable: 1 return specific port: 65535
03-31 19:16:25.504  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 39 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.504  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 40 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.505  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 127 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.666  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 30 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.666  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 33 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.666  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 35 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.666  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 36 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.666  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 37 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.666  1635  1640 I         : VSIP: configuration_impl:find_specific_port #1: service: 27 instance: 1 reliable: 1 return specific port: 65535
03-31 19:16:25.666  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 39 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.666  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 40 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.667  1635  1640 E         :** VSIP: Cannot find free client port for communication to service: 127 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.681  1635  2492 I         : VSIP: SUBSCRIBE(0000): [001f.0001.0001:ffff:1] true 1 accepted.
03-31 19:16:25.682  1635  1640 I         : VSIP: REMOTE SUBSCRIBE(0000): [001f.0001.0001] from 192.168.61.3:30491 unreliable was accepted
03-31 19:16:25.682  1635  1640 I         : VSIP: SUBSCRIBE ACK(3d1f): [001f.0001.0001.ffff]
03-31 19:16:25.922  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 30 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.923  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 33 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.923  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 35 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.923  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 36 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.923  1635  1640 E         : VSIP: Cannot find free client port for communication to service: 37 instance: 1 remote_port: 30501 reliable: 1
03-31 19:16:25.923  1635  1640 I         : VSIP: configuration_impl:find_specific_port #1: service: 27 instance: 1 reliable: 1 return specific port: 65535

haithem0501, 2023-05-22:
Thanks a lot for this. will be glad to check this.

haithem0501, 2023-05-22:
Thanks a lot for this. will be glad to check it.

goncaloalmeida, 2023-06-05:
Fix applied in the last release (3.3.8).
Please retest.

closed 2023-06-05 by goncaloalmeida

DiogoPedrozza, 2023-06-05:
More precisely the issue was fixed in the version [3.3.7](https://github.com/COVESA/vsomeip/releases/tag/3.3.7):
-"Release no longer in use client endpoints"

wjdqjqrnjs, 2023-09-20:
@goncaloalmeida 
Hello, I updated new issue #524 
Still new version (3.3.8),the same issue occurs.

Could I get help resolving the issue?
I wonder if the required conditions were not met when setting up or building in the new version.

Thank you!!